Qengineering/Jetson-Nano-Ubuntu-20-image #2: Does it work with the Jetson Nano 2GB?

davidakhihiero, 2021-11-10:
The image seems not to work with the Jetson Nano 2GB. It is stuck on the logo screen.

Qengineering, 2021-11-10:
I'm sorry. Indeed I made the image with 4 GB Jetson Nano.
Please follow the instruction in the website to upgrade manually.

lgkimjy, 2022-08-06:
I am currently using Jetson Nano 4GB, but I have the same issue that keeps stuck on the logo screen. 
Are there any problems expected??

Qengineering, 2022-08-06:
@lgkimjy,

Just flashed a fresh SD card with `JetsonNanoUb20_2.img.xz` downloaded from [Sync](https://ln5.sync.com/dl/741c98fe0/x8kxkhgs-cgmzk7rf-n4m7pyw8-h64tzbv5). It worked flawlessly. No problems.
First the logo screen, then some terminal commands and finally the 'regular' Nvidia screen.

merhovon, 2023-02-09:
I have used the mapping and unfortunately I have to notice that the performance is massively worse than with version 18. Currently the CPU is not overclocked, does it possibly make sense to extend the imaging so that the CPU is increased to 2.0?

Qengineering, 2023-02-09:
@merhovon,

On the 4 GB RAM version, there is no performance drop whatsoever.
On the 2GB RAM version, I don't know. Due to its limited amount, it makes sense that (time-consuming) memory swapping will occur more often than with the 4 GB version.
Overclocking the CPU can give you a few per cent more speed.