Qengineering/Jetson-Nano-Ubuntu-20-image #3: Unable to Get OS image from Gdrive Site

Zdravo1331, 2022-02-07:
This github page directs users to the Gdrive page to get the actual Ubuntu 20.0 OS for Nvidia Jetson Nano Dev Kit. But once at the site and trying to download the OS the following error appears: 
"Sorry, you can't view or download this file at this time.
Too many users have viewed or downloaded this file recently. Please try accessing the file again later. If the file you are trying to access is particularly large or is shared with many people, it may take up to 24 hours to be able to view or download the file. If you still can't access a file after 24 hours, contact your domain administrator." 

This error has consistently been appearing over the past 3 hours. 

Does anyone have a work around for this or can the file be placed here, on github?

Qengineering, 2022-02-08:
Due to reaching Gdrives' traffic limits, we've made a second download option available.

Zdravo1331, 2022-02-08:
Thank you! I was able to successfully download the image from the gdrive.

closed 2022-02-08 by Zdravo1331

FancyGentlemen, 2022-07-25:
I am having the same issue now.  The error happens with both links.  I'll try again tomorrow in the mean time incase too many people were downloading it in a 24 hour period.

lgkimjy, 2022-08-02:
Is this second download option includes PyTorch and opencv too?

Qengineering, 2022-08-03:
All downloads include PyTorch and OpenCV. Except the bare-bones images.