Qengineering/Jetson-Nano-Ubuntu-20-image #4: SD card size ?

ksirirux, 2022-02-15:
I use SDXC 128gb but after i flash system found it as 32GB ..
Can I solve this problem?

Thank you

Qengineering, 2022-02-15:
With a tool like GParted you can enlarge your partitions on the SD card.

ksirirux, 2022-02-16:
Thank you ... It's work!!

HusamZain, 2022-03-22:
@ksirirux  can explain how did you do it , or share a resources ?

thank you in advance

omersukruvarol, 2023-11-14:
Hi , I need help about same problem. I use 128gb sd card but after setup there is no enough free space

Qengineering, 2023-11-14:
With another PC and a tool like GParted you can modify your partitions on your SD  card. Find the /dev/mmcblk0p1 or something like that name (it will be ~ 30 GB) on your card. Enlarge that partition.

atmarachchige0081, 2025-01-20:
Use Mini Tool partition wizard to remove all the partitions of your SD card and you are good to go