Qengineering/Jetson-Nano-Ubuntu-20-image #6: performing ssh into jetson nano gives warnings

oscar-lima, 2022-02-21:
```/usr/bin/xauth:  file /home/jetson/.Xauthority does not exist```

oscar-lima, 2022-02-21:
solution:

On the jetson nano do:

    chmod 644 $HOME/.Xauthority

oscar-lima, 2022-02-22:
warning #2:

```X Error of failed request:  BadLength (poly request too large or internal Xlib length error) ...```

Qengineering, 2022-02-22:
Before jumping into the deep, is a dummy HDMI plug an option? See https://github.com/Qengineering/realsr-ncnn-Jetson-Nano/issues/4

oscar-lima, 2022-02-22:
Hello, thanks for the help. Actually I have attached a real hdmi display to the jetson, at least at setup phase. Later on I plan to remove it.

Qengineering, 2022-02-22:
OK, I'm very busy at the moment. I see what I can do this weekend.

oscar-lima, 2022-02-22:
Thanks, actually X11 works properly and does forward the display to the client via ssh. Is just an annoying warning but does not seem to affect the system.

Qengineering, 2022-02-26:
Strange.
Just flashed a fresh SD-card from the Gdrive.

SSH login : **jetson**
password : **jetson**

And this is the screen I get:
[image]
No warnings at all.
Even after `sudo apt-get update && sudo apt-get upgrade` with lots of new upgrades, no warnings.