Qengineering/Jetson-Nano-Ubuntu-20-image #9: Image creation question.

martinerk0, 2022-06-17:
Hi, can you describe workflow how to create such an image? I would like to create similar images for Xavier NX or other Jetson devices, and I used your image on nano successfully :)

Qengineering, 2022-06-17:
https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html

By the way, the latest JetPack 5.0 release has Ubuntu 20.04 on-board. (not available for the Nano yet)

longwm, 2022-06-17:
> 

If you download the Image which includes Pytorch + Open CV etc from the Gdrive link on Github it will run ROS2 no problem. I am using the image to run ROS2 Foxy without any issues. The image is 30 gigs once unzipped, you can write it using Balena etcher and then use Gparted to extend the partition to the full size of the SD card.

martinerk0, 2022-06-17:
@longwm yes but if you make changes to the OS and SD card, install some other packages, and then want to create smaller image (~10s of GBs) from larger SD card (100s of GBs) that contains lot of free space how do you do it? I saw some tutorials that you just dd it and zip should compress the free space but I wanted to know whole procedure from author ( that already has experience with it).

Qengineering, 2022-06-17:
If you're flashing an image from a 128GB card, balenaEtcher will want a 128GB card too, even if you're only using 24GB on your card. You need to reduce the size.
I used GParted to split a large partition (eg 128 GB) into two parts. One with the data (24 GB) and the other without any valid data (104 GB).
I then copied the 24 GB partition with `dd` to a new 32 GB SD card. (Some brands have only 31.3 GB of free space! [#16](https://github.com/Qengineering/Jetson-Nano-image/issues/16))
The next step is to use a tool like DiskInternals Linux Reader (if you have Windows) to create an image.
The image can now be zipped if you wish.

mrcmunir, 2022-08-20:
[jetcard-master.zip](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/files/9387853/jetcard-master.zip)
Hello I'm using this script for automatic autoresize needed Setup this python script into you sdcard before Create backup image.

Qengineering, 2022-08-21:
@mrcmunir ,

Try [option 2](https://github.com/NVIDIA-AI-IOT/jetcard#option-2---create-jetcard-from-scratch).

rwbot, 2023-02-05:
> @longwm yes but if you make changes to the OS and SD card, install some other packages, and then want to create smaller image (~10s of GBs) from larger SD card (100s of GBs) that contains lot of [...]

I'm interested in creating my own images as well, with different versions of ROS/Ubuntu for each. @longwm Did you figure out a solution?