Qengineering/Jetson-Nano-Ubuntu-20-image #10: Can I flash this to an eMMC module?

arjan-kuiper, 2022-07-03:
As the title states, I have a Jetson Nano module with 16GB of eMMC memory instead of an SD card slot. Is it somehow possible to flash this image into memory either with or without NVIDIA's SDKManager?

Qengineering, 2022-07-08:
I'm afraid, it's not possible. Best to follow the instructions at our site [Ubuntu 20.04](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html) to upgrade the OS manually.

arjan-kuiper, 2022-07-21:
> I'm afraid, it's not possible. Best to follow the instructions at our site [Ubuntu 20.04](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html) to upgrade the OS manually.

I've done like you said and it worked like a charm. Thank you very much!

closed 2022-07-21 by arjan-kuiper