Qengineering/Jetson-Nano-Ubuntu-20-image #11: No "tensorrt" module in python3.8 ?

Akiautiste, 2022-07-19:
dear author,i can not import python package:"tensorrt",is there not include tensorrt pybind in this ubuntu20.04 image?

Qengineering, 2022-07-21:
Unfortunately, it seems rather difficult to change the Python binding of TensorRT. I've been working on it for a while now, but no luck so far. TensorRT is an integral part of your JetPack and is not easy to install 'separately'. First of all, I don't see a solution to make TensorRT work with Python 3.8. If anyone has any suggestions, let me know!

Akiautiste, 2022-07-21:
> Unfortunately, it seems rather difficult to change the Python binding of TensorRT. I've been working on it for a while now, but no luck so far. TensorRT is an integral part of your JetPack and is [...]

haha, i solved it...
I extract ubuntu18.04 jetson nano image file,can find tensorrt packages in path:/usr/lib/python3.6/dist-packages/tensorrt ,
Just copy tensorrt*/ to ubuntu20.04,of course ,need build&install python3.6 in ubuntu20.04.

Qengineering, 2022-07-21:
Glad to hear you succeeded!

gagipro, 2022-08-09:
Hi,

got same issue and of course manage also to make it work with python 3.6.

But we need a way to have tensorrt with 3.8.

I see perhaps a solution with tiny tensorrt…

merhovon, 2023-02-09:
I was wondering why I get the following error message:

`I see perhaps a solution with tiny tensorrt… really not sure !`

Does anyone perhaps have a solution?

liutaiyuan1, 2023-04-10:
> haha, i solved it... I extract ubuntu18.04 jetson nano image file,can find tensorrt packages in path:/usr/lib/python3.6/dist-packages/tensorrt , Just copy tensorrt*/ to ubuntu20.04,of course ,need [...]

 I copied tensorrt*, but no, do you change anything else?

gagipro, 2023-04-10:
> I copied tensorrt*, but no, do you change anything else?

He managed with python 3.6...

liutaiyuan1, 2023-04-10:
> He managed with python 3.6...

Me too, but I can't use it

theikkila, 2023-06-15:
Found that you can pretty easily compile tensorrt from sources. I used the python3.8.10 shipped with the Image provided by Qengineering and followed these instructions to build tensorrt:

https://github.com/NVIDIA/TensorRT/tree/8.0.1#building-tensorrt-oss

```
cd $TRT_OSSPATH
mkdir -p build && cd build
cmake .. -DTRT_LIB_DIR=$TRT_LIBPATH -DTRT_OUT_DIR=`pwd`/out -DTRT_PLATFORM_ID=aarch64 -DCUDA_VERSION=10.2
make -j$(nproc)

sudo make install
```

I also created python bindings by following the instructions here:
https://github.com/NVIDIA/TensorRT/tree/8.0.1/python

closed 2023-07-13 by Qengineering