Qengineering/Jetson-Nano-Ubuntu-20-image #12: torchtext and torchdata

ketut, 2022-07-28:
I have problem when try implementation of https://nlp.seas.harvard.edu/annotated-transformer/ 
on torch 1.12.0 torchdata need torch 1.11.0, 
but on torch 1.11.0 torchtext got error 

''' >>> import torchtext
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jetson/.local/lib/python3.8/site-packages/torchtext/__init__.py", line 9, in <module>
    from . import vocab
  File "/home/jetson/.local/lib/python3.8/site-packages/torchtext/vocab/__init__.py", line 11, in <module>
    from .vocab_factory import (
  File "/home/jetson/.local/lib/python3.8/site-packages/torchtext/vocab/vocab_factory.py", line 4, in <module>
    from torchtext._torchtext import (
ImportError: /home/jetson/.local/lib/python3.8/site-packages/torchtext/_torchtext.so: undefined symbol: _ZNK5torch3jit6MethodclESt6vectorIN3c106IValueESaIS4_EERKSt13unordered_mapISsS4_St4hashISsESt8equal_toISsESaISt4pairIKSsS4_EEE '''

Qengineering, 2022-07-28:
@ketut ,

Did you install torchtext from scratch? As far as I know, there is no wheel for the aarch64 architecture.
see also [repo](https://gist.github.com/abishekmuthian/8d0ef5a3768c5d8d22e164863670330a)

ketut, 2022-07-30:
> @ketut ,
> 
> Did you install torchtext from scratch? As far as I know, there is no wheel for the aarch64 architecture. see also [repo](https://gist.github.com/abishekmuthian/8d0ef5a3768c5d8d22e164863670330a)

I have tried this, but it produces the same error

ketut, 2022-08-04:
Thank you @Qengineering, it's works

closed 2022-08-04 by ketut