Qengineering/Jetson-Nano-Ubuntu-20-image #13: Cannot import torch or torchvision on a different user

sergiodl-ubi, 2022-08-02:
Steps:

1. Created a new user
2. Added the same groups of user "jetson" to my new user 
3. `$ python3 -> import torch` returns `Illegal instruction (core dumped)`

torchvision returns the same error
Adding the new user to the "jetson" group doesn't help
There are no problems importing the modules from the "jetson" user

hakchul, 2023-03-16:
I've handled this problem with a simple solution.
Run below command for the added user.

`$ pip3 install numpy==1.18.5`

jasonacox, 2023-06-13:
Thanks @hakchul !   The "Illegal instruction" during `import torch` had me stumped...

```
jason@nano:~$ python3
Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
Illegal instruction (core dumped)
```

But after the fix `pip3 install numpy==1.18.5`:

```
jason@nano:~$ python3
Python 3.8.10 (default, Jun 22 2022, 20:18:18) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> torch.__version__
'1.12.0a0+git67ece03'

jason@nano:~$ wget --quiet https://raw.githubusercontent.com/jasonacox/ProtosAI/master/gpu/gpu.py
jason@nano:~$ python3 gpu.py 

GPU Information for PyTorch
   Version of torch:  1.12.0a0+git67ece03

GPU Details
   Device #0: NVIDIA Tegra X1
   Type: cuda 
   GPUs: 1

Memory
   Global Free Memory: 0.29264068603515625 GB
   GPU Memory: 3.871135711669922 GB
   Allocated: 0.0 GB
   Cached:    0.0 GB

PyTorch Test with cuda - Random 4x4 Array

tensor([[3984, 7897,  560,  696],
        [9498, 2317, 3607, 8440],
        [6648,  236, 7257, 1919],
        [3124, 7761, 7924, 4348]], device='cuda:0')
```