Qengineering/Jetson-Nano-Ubuntu-20-image #17: Tensorflow 2.5 already out for JetPack 4.6

hidoba, 2022-10-31:
There's a wheel for jetpack 4.6 of TF2.5 here https://elinux.org/Jetson_Zoo#ONNX_Runtime

Qengineering, 2022-11-01:
Are you sure?

I found a TF2.5 for Python3.6, not Ubuntu's 3.8.
Also found some ONNX_Runtime wheels intended for Python 3.8.

However, with a clear statement of the TensorFlow team that TF2.5 needs CUDA 11, I really doubt if it will work.
Sure, with a cross-compile you can generate everything you like. Did you test it?

muellevin, 2023-05-19:
@Qengineering it seems that nvidiais able to do it anyway. You are able to install tensorflow 2.7 on Jetpack 4.6 with the following steps:
```
# for tensorflow
sudo apt-get install libhdf5-serial-dev hdf5-tools libhdf5-dev zlib1g-dev zip libjpeg8-dev liblapack-dev libblas-dev gfortran -y
sudo apt-get install python3-pip -y
sudo python3 -m pip install --upgrade pip
sudo -H pip3 install testresources setuptools
# numpy .5 -> core dumb
sudo -H pip3 install numpy==1.19.4 future mock keras_preprocessing keras_applications gast protobuf pybind11 cython pkgconfig packaging
# h5py needs some special magic
sudo -H H5PY_SETUP_REQUIRES=0 pip3 install -U --no-build-isolation h5py==3.1.0
sudo -H pip3 install --extra-index-url https://developer.download.nvidia.com/compute/redist/jp/v461 tensorflow==2.7.0+nv22.1
```
Would it be possible after the upgrade to Ubuntu 20?

Qengineering, 2023-05-19:
@muellevin,

Are you sure the installed TensorFlow version uses CUDA? 
It is not the regular ARM - python 3.6 TensorFlow you have?

muellevin, 2023-05-20:
On current JP Version 4.6.3 with Ubuntu 18.04 after the installation i typed `python3 -c "import tensorflow as tf; print(tf.config.list_physical_devices('GPU'))` which returns `[PhysicalDevice(name='/physical_device:GPU:0', device_type='GPU')]`

If using the deprecated `tf.test.is_gpu_available()` it says that it found the NVIDIA Tegra X1 as GPU

The Tensorflow Version is from Nvidia. I was following their guide [guide](https://docs.nvidia.com/deeplearning/frameworks/install-tf-jetson-platform/index.html#verify) to install it but needed the modification from aove.