Qengineering/Jetson-Nano-Ubuntu-20-image #18: how to get CSI-Camera working on nanojetson-ubuntu 20.04

rodrigosalasm, 2022-12-16:
Dear all, first of all congratulations for the great work, it has been of great help in my project due to the ambiguous version of python that NVIDIA maintains in jetson nano, I would like to know how to make a CSI camera (raspberry or IMX) work with OpenCV in this system operational, follow this workflow https://github.com/JetsonHacksNano/CSI-Camera and got failure, stay tuned, regards.

Qengineering, 2022-12-16:
You have to use GStreamer of FFmpeg to get your RPi camera working. 
For instance,
```
$ gst-launch-1.0 nvarguscamerasrc sensor_mode=0 ! 'video/x-raw(memory:NVMM),width=3820, height=2464, framerate=21/1, format=NV12' ! nvvidconv flip-method=0 ! 'video/x-raw,width=960, height=616' ! nvvidconv ! nvegltransform ! nveglglessink -e
```
will give you a nice preview.
BTW, I'll be the first to admit that building a pipeline isn't easy.

rodrigosalasm, 2022-12-16:
Is there a way to do that in a python code using the OpenCV library.
Try this, I get error. (unable camera)

# MIT License
# Copyright (c) 2019-2022 JetsonHacks

# Using a CSI camera (such as the Raspberry Pi Version 2) connected to a
# NVIDIA Jetson Nano Developer Kit using OpenCV
# Drivers for the camera and OpenCV are included in the base image

import cv2

""" 
gstreamer_pipeline returns a GStreamer pipeline for capturing from the CSI camera
Flip the image by setting the flip_method (most common values: 0 and 2)
display_width and display_height determine the size of each camera pane in the window on the screen
Default 1920x1080 displayd in a 1/4 size window
"""

def gstreamer_pipeline(
    sensor_id=0,
    capture_width=1920,
    capture_height=1080,
    display_width=960,
    display_height=540,
    framerate=30,
    flip_method=0,
):
    return (
        "nvarguscamerasrc sensor-id=%d !"
        "video/x-raw(memory:NVMM), width=(int)%d, height=(int)%d, framerate=(fraction)%d/1 ! "
        "nvvidconv flip-method=%d ! "
        "video/x-raw, width=(int)%d, height=(int)%d, format=(string)BGRx ! "
        "videoconvert ! "
        "video/x-raw, format=(string)BGR ! appsink"
        % (
            sensor_id,
            capture_width,
            capture_height,
            framerate,
            flip_method,
            display_width,
            display_height,
        )
    )

def show_camera():
    window_title = "CSI Camera"

    # To flip the image, modify the flip_method parameter (0 and 2 are the most common)
    print(gstreamer_pipeline(flip_method=0))
    video_capture = cv2.VideoCapture(gstreamer_pipeline(flip_method=0), cv2.CAP_GSTREAMER)
    if video_capture.isOpened():
        try:
            window_handle = cv2.namedWindow(window_title, cv2.WINDOW_AUTOSIZE)
            while True:
                ret_val, frame = video_capture.read()

                if cv2.getWindowProperty(window_title, cv2.WND_PROP_AUTOSIZE) >= 0:
                    cv2.imshow(window_title, frame)
                else:
                    break 
                keyCode = cv2.waitKey(10) & 0xFF
                # Stop the program on the ESC key or 'q'
                if keyCode == 27 or keyCode == ord('q'):
                    break
        finally:
            video_capture.release()
            cv2.destroyAllWindows()
    else:
        print("Error: Unable to open camera")

if __name__ == "__main__":
    show_camera()
...................................................................................................
This code incorporates the pipeline and transfers the image format to bgr "OpenCV format", it does not open the camera anyway.

the error appears in the line: if video_capture.isOpened():

Thanks in advance for your quick attention.

Qengineering, 2022-12-16:
Right now, you are facing the pipeline nightmare.
It should work. Otherwise there is a typo in the code. Or just a parameter setting. You name it.
With almost no debug information from GStreamer, it can be a time consuming process.
As you have seen, the RPi camera is working in Ubuntu 20.04.
Unfortunately, I haven't got the time to debug code.

rodrigosalasm, 2022-12-16:
ok, thanks for your time

mrceki, 2023-01-28:
@rodrigosalasm which cam are you using?

mrceki, 2023-06-01:
After struggling for a while, I gave up and realized that I needed a wide-angle camera so bought a usb camera. @rahulswa08

rahulswa08, 2023-06-01:
Ohh which camera are you using? @mrceki

mrceki, 2023-06-01:
> Ohh which camera are you using? @mrceki

IMX179 8MP USB Camera (A)

rahulswa08, 2023-06-01:
Thanks @mrceki . I was using PiCamera v1 and when I changed to PiCamera v2 the issue seemed to be resolved.