Qengineering/Jetson-Nano-Ubuntu-20-image #19: Will this work for Jetson TX2?

fpm-github, 2023-01-20:


Qengineering, 2023-01-20:
I don't know.
You can try to install Ubuntu 20.04 manually following [our guide](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html). 
I never tried it on a TX2, so it will be an adventure.
Please let me (and the other readers) know how it went, if you decide to step into the deep.

fpm-github, 2023-01-20:
I 'm new using Jetson Tx2.
Any recommendations or guide you can provide please for installing Python 3.8.10v, Pytorch 1.12.0v and TorchVision 0.13.0v for JetPack 4.6.1v (Ubuntu 18.04)?

Qengineering, 2023-01-20:
Dear @fpm-github,

Before you can install Pytorch and TochVision, you need indeed Python 3.8, which, on its turn, needs Ubuntu 20.04.
JetPack 4.6 ships with Ubuntu 18.04 and Python 3.6.
The only way to get all things working is by upgrading Ubuntu 18 to 20.04.
You could try to upgrade only Python. However, that is something I don't recommend. Python runs at the hart of your system, and two versions simultaneous running is asking for disaster.
Install your JetPack 4.6. See if everything is working well. Try to upgrade to Ubuntu 20 with the guide. Next, install the remaining software.

aniruddha236, 2023-12-11:
@fpm-github were you able to upgrade to Ubuntu 20.04 on the TX2 using the guide from @Qengineering? I will be attempting to do the same in the following days and would update regarding the same.

Previously, what worked for us was to create a Python 3.8 virtual environment and then install Pytorch and Torchvision. Our goal was to run YOLO object detection on the Jetson Nano and [this link](https://i7y.org/en/yolov8-on-jetson-nano/) was very helpful. With this approach, we didn't have to upgrade to Ubuntu 20.04.

However, now we are interested in running ROS2 applications and would like to use Ubuntu 20.04 on a old Jetson TX2.