Qengineering/Jetson-Nano-Ubuntu-20-image #21: I can't load

Filmimagine, 2023-02-13:
I can't load Ubuntu 20.04 image, with OpenCV 4.5.3, TensorFlow 2.4.1 and PyTorch 1.9.0.

Qengineering, 2023-02-13:
@Filmimagine ,

Where is it going wrong?
When you have flashed the SD card and put it in the Jetson Nano, what do you see on the screen?
By the way, do you have a Jetson Nano 4 GB RAM?
Other versions are not supported.

Filmimagine, 2023-02-14:
Yes, I have a Jetson Nano 4 GB when I load an Ubuntu 20.04 image with OpenCV 4.5.3, TensorFlow 2.4.1 and PyTorch 1.9.0 onto my computer to flash the SD card during the download. NETWORK_FAILED which I think might be my internet but i tried changing internet and its still the same but i can load JetsonNanoUb20_2.img.xz

Qengineering, 2023-02-14:
@Filmimagine ,

I'm not sure if I understand it right.
First you need the download the JetsonNanoUb20_2.img.xz onto your computer is a folder. Say the folder "Downloads".
Once the download is fully completed, you can start flashing the SD-card. It are two separate things. You can't download and at the same time burn the card.

Filmimagine, 2023-02-15:
@Qengineering 
Now i can do it. i used sd card 64 gb but after i flashed system found it was 32gb.
How can I solve this problem?
thank you

Qengineering, 2023-02-16:
@Filmimagine,

That's normal behavior. The SD burner allocates a 32 GB partition on the card and writes its content byte by byte.
The other 32 GB (in case of a 64 GB SD-card) will remain unused (un-allocated).

You can use tools like Gparted to expand the original 32 GB partition with the other unused 32 GB.
All the above suit be done on a separate Linux machine. 
You cannot use the Jetson Nano to expand its own primary partition.