Qengineering/Jetson-Nano-Ubuntu-20-image #23: tensorflow update

hannesa2, 2023-03-12:
I tried

```
sudo apt install python-is-python3
sudo apt install python3-venv
rm -rf venv/
python -m venv venv
source venv/bin/activate
python -m pip install --upgrade pip
```

and tried
`pip install tensorflow==2.10.1`
but
`python -c 'import tensorflow as tf; print(tf.__version__)'`
show me 

> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/__init__.py", line 37, in <module>
>     from tensorflow.python.tools import module_util as _module_util
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/__init__.py", line 45, in <module>
>     from tensorflow.python.feature_column import feature_column_lib as feature_column
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/feature_column/feature_column_lib.py", line 18, in <module>
>     from tensorflow.python.feature_column.feature_column import *
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/feature_column/feature_column.py", line 143, in <module>
>     from tensorflow.python.layers import base
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/layers/base.py", line 16, in <module>
>     from tensorflow.python.keras.legacy_tf_layers import base
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/keras/__init__.py", line 25, in <module>
>     from tensorflow.python.keras import models
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/keras/models.py", line 22, in <module>
>     from tensorflow.python.keras.engine import functional
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/keras/engine/functional.py", line 32, in <module>
>     from tensorflow.python.keras.engine import training as training_lib
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/keras/engine/training.py", line 54, in <module>
>     from tensorflow.python.keras.saving import hdf5_format
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/keras/saving/hdf5_format.py", line 37, in <module>
>     import h5py
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/h5py/__init__.py", line 46, in <module>
>     from ._conv import register_converters as _register_converters
>   File "h5py/_conv.pyx", line 1, in init h5py._conv
>   File "h5py/h5t.pyx", line 293, in init h5py.h5t
>   File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/numpy/__init__.py", line 320, in __getattr__
>     raise AttributeError("module {!r} has no attribute "
> AttributeError: module 'numpy' has no attribute 'typeDict'

so I did
`pip install numpy==1.20`
but `python -c 'import tensorflow as tf; print(tf.__version__)'`

RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xe
ImportError: numpy.core.multiarray failed to import

The above exception was the direct cause of the following exception:

SystemError: <built-in method __contains__ of dict object at 0x7f7034b640> returned a result with an error set

The above exception was the direct cause of the following exception:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/__init__.py", line 37, in <module>
    from tensorflow.python.tools import module_util as _module_util
  File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/__init__.py", line 37, in <module>
    from tensorflow.python.eager import context
  File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/eager/context.py", line 35, in <module>
    from tensorflow.python.client import pywrap_tf_session
  File "/home/hannes/CarND-Behavioral-Cloning-P3/venv/lib/python3.8/site-packages/tensorflow/python/client/pywrap_tf_session.py", line 19, in <module>
    from tensorflow.python.client._pywrap_tf_session import *
ImportError: initialization failed
```

Does someone knows, how to fix this ?

Qengineering, 2023-03-12:
TensorFlow 2.5 and higher depend all on CUDA 11.0 and cuDNN version 8.0.4, both not yet available for the Jetson Nano.
Jetson Nano JetPack 4.6.1 and this version (Ubuntu 20.04), both have CUDA 10.2.
Due to some low-level GPU drivers, upgrading CUDA is almost impossible.

hannesa2, 2023-03-12:
Thanks for clarification

closed 2023-03-12 by hannesa2