Qengineering/Jetson-Nano-Ubuntu-20-image #25: Udisk Image？

MagicalBrain, 2023-03-22:
My board with jetson nano does not have  any slot of sd card, so I need an ubuntu 20.04 image of USD flash disk. Can I get support?

Qengineering, 2023-03-22:
Do you mean: boot from **USB** disk?
I have been unable to boot Ubuntu 20.04 from USB so far. Some low-level drivers throw a spanner in the works.

AndresRoc, 2023-03-24:
Does that mean it works? I want to boot 20.04 on a JetsonNano from Waveshare, which has a micro-sd slot, but after a lot of tries discovered that it does not work

Qengineering, 2023-03-25:
Booting from USB is **not** possible. The way it works for the original Tegra OS (Ubuntu 18.04) OS, fails to work on an Ubuntu 20.04 OS. Perhaps someone out there in the field has the solution, but I didn't get working.
Once it boots there are no other HW communication issue as far as I know.

MagicalBrain, 2023-05-24:
@Qengineering Finally, I find the way to solve this problem: I can follow your [guidance](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html) to upgrade ubuntu 18.04 to ubuntu20.04 on offical Udisk image. It does work, but it take me to much time.

closed 2023-05-24 by MagicalBrain