Qengineering/Jetson-Nano-Ubuntu-20-image #28: Wifi Issue

Gavman04, 2023-03-29:
We are flashing Jetson Nano 4gb with this image and using Wifi Card: Intel Model 8265NGW. We periodically have an error where the Jetson Nano does not recognize the wifi adaptor. No network settings show up at all and it doesn't not connect to any of its saved wifi networks (they don't show up at all.) To test the issue we updated everything all of the software and then rebooted 10 times. 2/10 times the Jetson had this issue still. We're not sure what's going on here. Thanks for your help!

Qengineering, 2023-04-26:
First of all, apologies for the delay in response. At the moment, the company is busy, which results in a delay in handling issues.

Unfortunately, we do not have any WiFi module for the Jetson Nano. It makes testing difficult.
I must say that the idea of [kayccc](https://forums.developer.nvidia.com/t/jetson-nano-wifi-adaptor-not-found-intel-8265ngw/247895) was not that bad at all. See if you also get these errors with the original JetPack. Given the number of posts on the internet, I think it's more related to the Jetson Nano than Ubuntu 20.04. See for instance https://forums.developer.nvidia.com/t/jetson-nano-intel-8265-wifi-isn-rsquo-t-recognized/76808/15

I'm sorry I can't help you further.

mtmal, 2023-09-10:
Hi all,

I had similar issue with my nano. With the original JetPack, my intel WiFi was working flawlessly, but this wasn't a case in this version. This post provides a solution which, at least for now, is successful on my end: https://forums.developer.nvidia.com/t/wifi-m-2-intel-3168ngw-loading-but-can-not-use-it/72037/15

In a nutshell, you need to append pcie_aspm=off to kernel command line in /boot/extlinux/extlinux.conf

Zelo-S, 2023-09-12:
> 

mtmal's solution worked for me too : )