Qengineering/Jetson-Nano-Ubuntu-20-image #31: Falled to start Load Kernel Modules

newr66, 2023-05-02:
Hello
There are problems when loading the kernel.
Perhaps you can help solve it.
Tried installing the last two images, the result is the same. Used 32 GB and 64 GB SD card.
The only thing is probably that Jetson Nano is for Chinese market. 
[image]

Qengineering, 2023-05-02:
@newr66,
I'm sorry to hear about your issue.
I think is going to be hard, even impossible to solve.
https://forums.developer.nvidia.com/t/insmod-issue-disagrees-about-version-of-symbol-module-layout/225896/4
https://stackoverflow.com/questions/2720177/module-layout-version-incompatibility
Best, you start with an original JetPack 4.6.3 and use [the guide](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html) to upgrade to Ubuntu yourself. In that case, you are always sure the hardware matches the software.

newr66, 2023-05-08:
Thank you for your reply.
Installed Ubutu 20.04 but cannot build Opencv with Cuda. The installation is crashing.
At the same time on 18.04 I was building.
Used your build script(https://github.com/Qengineering/Install-OpenCV-Jetson-Nano/raw/main/OpenCV-4-5-1.sh), same situation.
I will attach the [CMakeError. ](https://drive.google.com/file/d/19sseVOtLJ5jtnGv32oSzMgZffl17DPfA/view?usp=share_link)

Qengineering, 2023-05-08:
@newr66 ,

Probably, you haven't install all the required software dependencies.
OpenCV complains about a missing /sys/video.h file (near the end of CMakeError.log)
`sudo apt-get install -y libv4l-dev v4l-utils qv4l2` will solve the issue.

closed 2023-05-08 by newr66