Qengineering/Jetson-Nano-Ubuntu-20-image #32: can I install it in external SSD and run it from there ?

RamaManohar5, 2023-05-03:
can I install it in external SSD and run it from there ?

Qengineering, 2023-05-03:
Unfortunately not. I've tried several different approaches and never succeeded. 
Please let me know if you have more luck

RamaManohar5, 2023-05-05:
Hi, thanks for the response. I've found the solution to run this ubunut 20.04 image from SSD. we just need to copy data to SSD using bootfromUSB and then mount it. we have to make changes in SD card boot extlinux.conf (set SSD partuuid as root). and don't remove the SD card. I am able to boot from SSD.

podfrogs, 2023-06-01:
I am running on a USB stick. Requires a few tricky configurations and reboots. 
1) After connecting the USB stick, create fs and mount it (say /mnt/usb1). Then update etc/fstab (on the SD card which is mounted as /) to automatically mount the USB stick. sudo blkid can be used to get the UUID for this. Also not the device (example: /dev/sda1) Then Reboot.
2) The USB stick should be automatically mounted to mnt/usb1. At this point, run sudo rsync -axv / /mnt/usb1 
that copies all files from SD card to the USB stick
After the copy completes, reboot. There may be errors, ignore and reboot. 
3) After rebooting, repeat the command again sudo rsync -axv / /mnt/usb1 . Any remaining file will be incrementally copied. At this point, there should no errors. 
Edit /boot/extlinux/extlinux.conf 
Copy all lines under LABEL primary into another section and rename it as LABEL sdcard
Edit the APPEND line on the "primary" section and change the root device to root=/dev/sda1 (instead of sdcard root=/dev/mmcblk0p1)
Save. DO NOT REBOOT at this time.
Edit /etc/fstab and mount the USB to / and mount the SD Card to /mnt/sdcard (make sure this directory exists)
Save. DO NOT REBOOT at this time. Check fstab with sudo findmnt --verify
If no errors, copy /boot/extlinux/extlinux.conf  to /mnt/usb1/boot/extlinux/extlinux.conf
and also copy /etc/fstab to /mnt/usb1/etc/fstab
At this point, the extlinux.conf & fstable on SD Card and USB will be the same. 
Reboot.
Jetson Nano will run on usb. Check the mounts. USB will be mounted to / and SD card will be mounted to /mnt/sdcard . You may rsync files directly from /mnt/sdcard to another USB if you want to change the drive

trx1138, 2023-07-23:
@RamaManohar5 @Qengineering 
> Unfortunately not. I've tried several different approaches and never succeeded. Please let me know if you have more luck

I made it to boot straight to a usb without a sdcard.
following is just a log of what I did to make it work and ended repeated trials. some of these could be irrelevant.

1. flash pre-built image TWICE. once to the usb drive and another to the _sdcard_.
2. boot to _sdcard_ then format usb root(`/dev/sda1`) with ext4, write down uuid of the newly formatted patition then mount. (`/mnt/usb1`)
3. rsync clone _sdcard_'s root to usb like @podfrogs  mentioned. reboot and clone again to make sure.  `>> sudo rsync -axv / /mnt/usb1`
4. edit `/mnt/usb1/extlinux.conf` like the example in this post. [https://collabnix.com/how-to-boot-nvidia-jetson-nano-from-usb-instead-of-sd-card/](https://collabnix.com/how-to-boot-nvidia-jetson-nano-from-usb-instead-of-sd-card/)
5. remove _sdcard_ then reboot. voila!
 
interesting part which draws attention is `extlinux.conf` edit in step #4.
there's TWO entry blocks, one labeled as _primary_ with uuid and another labeled as _sdcard_ with device path but in fact both entries are pointing the same device. and it's a bit weird too to make it default to boot sdcard, but... anyways it works.

I'm suspecting #1 and #4 could be the key elements which affected the result but not sure.

ps. thanks for great works including this prebuilt img. got helped many times so far. ;D

replicatesyndicate, 2023-10-31:
One convenient approach that I've found modifies @trx1138 's method:
- Flash the jetson image from this repository directly onto a USB drive using Balena or any other tool.
- `$ sudo blkid`to get USB PARTUUID. Find the 32 GB partition's `/dev/sd**` address on the USB drive, and copy the associated PARTUUID. Copy that PARTUUID into `boot/extlinux/extlinux.conf` on the drive.
- Unmount the drive, and plug it into your Jetson Nano with the power off.
- Remove the SD card on the Jetson Nano before turning it on with the USB drive.
- Reattach the SD card if you wish. You can also use GParted on it with ease now.

gth-42, 2023-12-22:
Another take on the instructions: 

"Flash" to device:  
I use Ubuntu's Startup Disk Creator {usb-creator-gtk}, or equivalent.  No need for pre-partitioning, formatting, or other rituals.  SDC creates a GPT partition and dumps the contents correctly regardless of what's on the drive. 

Access the newly created filesystem from the same host.  This assumes a properly configured automounter and file browser.  You may need to make yourself owner or find another way get RW permissions.  I use: 
`sudo chown -R yourname /media/yourname/*`

From the mount point (or removable drive in browser), open /boot/extlinux/extlinux.conf in a text-ONLY editor (e.g. mousepad).  Make a copy of that whole "LABEL primary" block.  Change the LABEL on one of them to "backup" or equivalent. 
What we need is the *partition* UUID rather than the one for the disk.  "Disks", to my knowledge, doesn't give you the PARTUUID.  I use:
`blkid /dev/sdb1`
copy the whole "PARTUUID="..." block, and paste it next to "root=" in place of /dev/mmcblk0p1, then remove the quotes.  My resulting file looks something like:

TIMEOUT 30
DEFAULT primary

MENU TITLE L4T boot options

LABEL primary
      MENU LABEL primary kernel
      LINUX /boot/Image
      INITRD /boot/initrd
      APPEND ${cbootargs} quiet root=PARTUUID=5ac8Od7c-40fb-47x6-bd56-4110q389819b rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 

LABEL backup
      MENU LABEL primary kernel
      LINUX /boot/Image
      INITRD /boot/initrd
      APPEND ${cbootargs} quiet root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0 

Save the file, safely eject, and hang on to your hat.  1TB NVME in a USB enclosure works great with a proper-spec cable.  I haven't tried a key adapter for the slot under the card -- yet.

Optionally, resize the partition prior to first boot while the drive is still plugged in to the host.

If extlinux.conf is still read-only, figure out how to get permissions.  I suspect the big kicker with online "tutorials" is confusion about drive vs partition UUID and syntax.  

I'm not sure about jetpack version 4.5+ and flashing the "QSPI-NOR" first. Trying Nvidia's image with 4.6.x first (going through setup), getting irritated, and trying @Qengineering's image with a change to extlinux.conf worked great for me.  I don't know the state of my QSPI-NOR prior to the aforementioned.

mitchross, 2024-02-29:
> Another take on the instructions:
> 
> "Flash" to device: I use Ubuntu's Startup Disk Creator {usb-creator-gtk}, or equivalent. No need for pre-partitioning, formatting, or other rituals. SDC creates a GPT partition and dumps the contents correctly regardless of what's on the drive.
> 
> To access the newly created filesystem from the same host, you may need to make yourself owner or find another way get RW permissions. This is where plugging the drive into a working Nano (e.g. w/SD) might be helpful in some cases.
> 
> From the mount point (or removable drive in browser), open /boot/extlinux/extlinux.conf in a text-ONLY editor (e.g. mousepad). Make a copy of that whole "LABEL primary" block. Change the LABEL on one of them to "backup" or equivalent. What we need is the _partition_ UUID rather than the one for the disk. "Disks", to my knowledge, doesn't give you the PARTUUID. I use: `blkid /dev/sdb1` copy the whole "PARTUUID="..." block, and paste it next to "root=" in place of /dev/mmcblk0p1, then remove the quotes. My resulting file looks something like:
> 
> TIMEOUT 30 DEFAULT primary
> 
> MENU TITLE L4T boot options
> 
> LABEL primary MENU LABEL primary kernel LINUX /boot/Image INITRD /boot/initrd APPEND ${cbootargs} quiet root=PARTUUID=5ac8Od7c-40fb-47x6-bd56-4110q389819b rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0
> 
> LABEL backup MENU LABEL primary kernel LINUX /boot/Image INITRD /boot/initrd APPEND ${cbootargs} quiet root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 console=ttyS0,115200n8 console=tty0 fbcon=map:0 net.ifnames=0
> 
> Save the file, safely eject, and hang on to your hat. 1TB NVME in a USB enclosure works great with a proper-spec cable. I haven't tried a key adapter for the slot under the card -- yet.
> 
> Optionally, resize the partition prior to first boot while the drive is still plugged in to the host.
> 
> If extlinux.conf is still read-only, figure out how to get permissions. I suspect the big kicker with online "tutorials" is confusion about drive vs partition UUID and syntax.
> 
> I'm not sure about jetpack version 4.5+ and flashing the "QSPI-NOR" first. Trying Nvidia's image with 4.6.x first (going through setup), getting irritated, and trying @Qengineering's image with a change to extlinux.conf worked great for me. I don't know the state of my QSPI-NOR prior to the aforementioned.

this worked great!

Thortergill, 2024-04-20:
> Another take on the instructions:
>
> "Flash" to device: I use Ubuntu's Startup Disk Creator {usb-creator-gtk}, or equivalent. No need for pre-partitioning, formatting, or other rituals. SDC creates a [...]

Excuse my ignorance, but are you editing the extlinux.conf file that is on the original Nano, or the one on the new external SSD?

Many thanks!

logistc, 2025-03-24:
**The problem keys are from mciroSD Cards(including USB3.0 booted SSD).** The important point is to modify the emmc extlinux.conf file according to the different mciroSD Cards. Different mciroSD Cards need different settings for the emmc boot/extlinux/extlinux.conf file.

I've tried many times. The findings as follows:

1. SAMSUNG EVO Plus 128GB U3 A2 V30, the emmc extlinux.conf modify to 'root=/dev/mmcblk1p1'
2. SanDisk Ultra U1 A1 32GB,  the emmc extlinux.conf file modify to 'root=/dev/sda1' and so on. If you plug thress mciroSD Cards of SanDisk Ultra U1 A1 32GB into three USB card readers on jetson nano, you will find the results of command 'df -h' are /dev/mmcblk0p1, /dev/sda1, /dev/sdb1, /dev/sdc1. Because 'sd' is short for the brand  SanDisk. 'sda1' is the first mciroSD Cards of SanDisk Ultra U1 A1 32GB, 'sdb1' is the second mciroSD Cards of SanDisk Ultra U1 A1 32GB, 'sdc1' is the third mciroSD Cards of SanDisk Ultra U1 A1 32GB.
3. USB3.0 booted SSD refer to the comments of [robotasun](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/32#issuecomment-1787195432)
4. If you don't know the mciroSD Cards type and PARTUUID, please check and record the mciroSD Cards type and PARTUUID using 'lsblk' or 'blkid' or 'df -h' before flashing the image on the SD card.

That's all! Hope this helps, good luck!