Qengineering/Jetson-Nano-Ubuntu-20-image #34: Is there a way to install qt4

emresagir, 2023-05-21:
I'm sorry that my question is little bit off but couldn't find a solution.

How to Install qt4 on Jetson Nano (ubuntu 20.04) for Hector Slam

Firstly, I tried

`sudo add-apt-repository ppa:rock-core/qt4`

this code and tried to install the needed libraries, but in my opinion it couldn't find the libraries again because of the jetson nano's architecture.

Then I tried to download the debian packages from the internet and realized most of them are amd64 packages, I find one package for arm64, but this one needs dependencys as shown below

[tried to install downloaded package for 20.04 and arm64](https://i.stack.imgur.com/RIgyk.png)

When this gives an error, I tried to download the qt5 then change the version manually, as shown below it doesn't work and says run make install (don't know how to do)

[qt version 4.8](https://i.stack.imgur.com/uL2nE.png)

I need to install qt4 to run Hector Slam. What should I do any suggestions?

Qengineering, 2023-05-22:
@emresagir,

Nine times out of ten, the only way to install software on an ARM-based board is to build it from scratch. 
In the case of Qt, it certainly is, as you noticed.

I think [this guide](https://het.as.utexas.edu/HET/Software/html/qt-embedded-install.html) from the Qt development team can help you a lot. I haven't tested it, but it certainly looks like a good starting point.

Two hints. 
Firts, remove all old Qt installations before starting. It can be a nightmare if, for example, two library versions with the same name exist.
Second, you need a lot of memory. Increase your swap space to at least 4 or 6 GB. On [this page}(https://qengineering.eu/install-opencv-4.5-on-jetson-nano.html) you will find a way to do that.

emresagir, 2023-05-23:
Thanks for your answer, if I am right, the guide you recommend is explains cross-compiling but I want to build the qt in the jetson nano without needing for the cross-compiling. Is there a way to do host-host compile or I must do it that way, I couldn't figure out.

emresagir, 2023-05-24:
I accidentally solved the problem, now I can run hector slam without the qt4 warning, I don't know which step solved it but after the deadline of our project I will restart the process to find the solution, thanks for your help.

closed 2023-05-24 by emresagir