Qengineering/Jetson-Nano-Ubuntu-20-image #35: pip install -r requirements.txt stuck at "Building wheels for collected packages: grpcio"

podfrogs, 2023-06-01:
When running pip install -r requirements.txt it is stuck at 
"Building wheels for collected packages: grpcio"  
Is this normal? 

Building wheels for collected packages: grpcio
  Building wheel for grpcio (setup.py) ... /
I am using the unclocked version, after an apt-get upgrade
Thanks

Qengineering, 2023-06-02:
Which software package are you installing, containing the mentioned requirement.txt ?
I like to reproduce your issue.

benjmerales, 2023-10-23:
Hi. For future references, it's not stuck.

There are times when the PyPi server is slow, or maybe your internet is unstable.

There are also times where the wheel takes too long to build, especially when C compiling is involved.

To check the full logs of the installation, try running with the verbose keyword:
`pip install -vvv -r requirements.txt`

I had a similar case when installing PyQt5 that took almost 2 hours.