Qengineering/Jetson-Nano-Ubuntu-20-image #37: Connection to Pixhawk

oki-aryawan, 2023-07-11:
I'm trying to connect my Jetson Nano using your OS (Ubuntu 20.04) and I tried this tutorial:
https://www.hackster.io/Matchstic/connecting-pixhawk-to-raspberry-pi-and-nvidia-jetson-b263a7

and I got error when tried to execute:
`sudo mavproxy.py --master=/dev/ttyTHS1`
come up with this error:
`Illegal instruction`

can you help me with this one? Thank you

Qengineering, 2023-07-11:
@oki-aryawan,

Sorry to hear about your problem.
Looking at the Matt Clarkes project, I'm nor sure how I can help you.
It's all about a serial interface running on a Jetson Nano. 
It is going to be a time-consuming, low-level debug session to investigate the issue.
With not much to go on. 
I would suggest to contact Matt via Hackster. He knows best what can be the cause.

oki-aryawan, 2023-07-11:
okay, thank you

ItayShiff, 2023-11-16:
Did you find any solution ?