Qengineering/Jetson-Nano-Ubuntu-20-image #42: Automatically upgraded to ubuntu 22.04 (jammy) when accepting firefox update

Ahsan-khan, 2023-07-29:
Hello, I have tried to update my ubuntu on my nano from 18 to 20 but during the upgrading there was a prompt asking me to accept firefox update. I did and now my ubuntu is 22.04 and the nano doesn't work.

What should I do?

Qengineering, 2023-07-30:
@Ahsan-khan,

Sorry, do cure so far. Return to Ubuntu 20.04.
(I've tried your journey several times, but didn't succeed.)

nguyenhunga5, 2023-08-03:
@Qengineering I using firmware from Nvidia and updated to Ubuntu 22, I using "GNOME Flashback Desktop", but still have some issues (sleep, ...)

nguyenhunga5, 2023-08-04:
Update: I using your image (20.4) and upgrade to 22.4, after finish upgrade, it cannot login. I install Mate desktop and switch to it, now it working.
[image]

Qengineering, 2023-08-04:
@nguyenhunga5 ,

That looks great! Congratulations! 
Please tell me after a month of use, if everything still works to your expectations.

KalanaRatnayake, 2024-05-06:
Any update on this @nguyenhunga5 
How did the Nano perform? Did the GPU work?

KalanaRatnayake, 2025-05-21:
@sbarcelona11 

Can you check whether you can compile the CUDA examples and run them? 

If i remember, I also tried to upgrade to 22.04 but had to park it because there was a gcc incompatibility cant remember what. (it could be that gcc version in 22.04 was too high to compile cuda 10.2 which is the latest supported on nano)

sbarcelona11, 2025-05-21:
@KalanaRatnayake Can you check something like this:

sudo apt install gcc-7 g++-7 gcc-9 g++-9 gcc-10 g++-10

sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 70
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-9 90
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-10 100

sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-7 70
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-9 90
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-10 100

Select the version that you need:
sudo update-alternatives --config gcc