Qengineering/Jetson-Nano-Ubuntu-20-image #43: Teamviewer cant login

drjajas, 2023-08-11:
After starting my jetson nano with the new image i tried to connect Teamwiever with my acc. but i only get error:

"
There was a connectivity issue. Please check your internet connection and try
again.
"

I reinstalled and checked my Network but nothing worked.

Qengineering, 2023-08-11:
@drjajas ,

TeamViewer  can be really annoying! It is very hard to debug. It can be a Jetson Nano issue, or a PC problem or a network problem. You name it. Previous, I never had any problem with TeamViewer, but lately it cumbersome. 
I'm sorry.

drjajas, 2023-08-11:
i got it to work. i dont know which command did it but here is my history:

apt-get remove --purge teamviewer-host
sudo apt-get remove --purge teamviewer-host
sudo apt-get autoremove
rm -rf ~/.teamviewer
sudo rm -rf /var/log/teamviewer*
sudo find / -name "*teamviewer*" 2>/dev/null
sudo apt-get clean
sudo apt-get update
sudo reboot
sudo dpkg -i teamviewer-host_15.4.4445_armhf.deb
sudo dpkg -i teamviewer-host_armhf.deb
sudo apt install -f
sudo teamviewer setup
sudo systemctl enable teamviewerd.service
sudo apt install ./teamviewer-host_arm64.deb
sudo apt --fix-broken install
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json /usr/share/glvnd/egl_vendor.d/50_mesa.json.backup
sudo rm /usr/share/glvnd/egl_vendor.d/50_mesa.json
sudo apt-get install -f
sudo rm /var/cache/apt/archives/libegl-mesa0_21.2.6-0ubuntu0.1~20.04.2_armhf.deb
sudo apt-get update
sudo apt install ./teamviewer-host_arm64.deb
sudo teamviewer setup

drjajas, 2023-08-11:
I have no idea Why but it worked after i did this: 

sudo apt-get remove --purge teamviewer-host
sudo apt-get autoremove
rm -rf ~/.teamviewer
sudo rm -rf /var/log/teamviewer*
sudo find / -name "*teamviewer*" 2>/dev/null
sudo apt-get clean
sudo apt-get update
sudo reboot
sudo dpkg -i teamviewer-host_armhf.deb
sudo apt install -f
sudo teamviewer setup
sudo systemctl enable teamviewerd.service
sudo apt install ./teamviewer-host_arm64.deb
sudo apt --fix-broken install
sudo cp /usr/share/glvnd/egl_vendor.d/50_mesa.json /usr/share/glvnd/egl_vendor.d/50_mesa.json.backup
sudo rm /usr/share/glvnd/egl_vendor.d/50_mesa.json
sudo apt-get install -f
sudo rm /var/cache/apt/archives/libegl-mesa0_21.2.6-0ubuntu0.1~20.04.2_armhf.deb
sudo apt-get update
sudo apt install ./teamviewer-host_arm64.deb
sudo teamviewer setup

if it is working by you too dont quit it when authenticating your E-Mail  couse than it wont work again. Maybe it helps someone.

Qengineering, 2023-08-11:
@drjajas,

Thanks! I keep the issue open and label it as "Valued information"

drjajas, 2023-08-14:
Seems like it was only a short fix after a reboot it stopped working again.

Qengineering, 2023-08-14:
Too bad

drjajas, 2023-08-14:
A new way i found to fix it: 

wget https://download.teamviewer.com/download/linux/version_14x/teamviewer-host_armhf.deb
sudo apt-get remove --purge teamviewer-host
sudo dpkg -i teamviewer-host_armhf.deb
sudo apt install -f
sudo apt --fix-broken install
sudo dpkg -i teamviewer-host_armhf.deb
mkdir new
cd new/
wget https://download.teamviewer.com/download/linux/teamviewer-host_armhf.deb
sudo dpkg -i teamviewer-host_armhf.deb
sudo apt --fix-broken install
sudo dpkg -i teamviewer-host_armhf.deb
sudo teamviewer setup

i keep you updated if this also stops working.