Qengineering/Jetson-Nano-Ubuntu-20-image #44: Can not download image from sync.com

mishabeliy15, 2023-08-17:
Hey!
I'm very glad to see your image that already have everything I need.
But I can't download it.

https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/assets/25367184/767c38e8-1ee9-4914-bd38-45459ee94204

Qengineering, 2023-08-17:
@mishabeliy15,

I noticed it also. I don't know what causes the download to fail. I worked the last months without any problem.
The only thing I can imagine is the number of downloads allowed. At this moment 200 visitors are downloading simultaneously the file. Please try it at a later moment.
If the issue persist, I will take proper action.

mishabeliy15, 2023-08-17:
@Qengineering ,
I also can't download any partition of the splited image.

Qengineering, 2023-08-18:
@mishabeliy15,

The link is working again. Let's assume it was the Sync server down for a moment.
Please contact if your still having trouble downloading

closed 2023-08-18 by Qengineering

Xilient, 2024-02-17:
I still having trouble downloading. Can you provide an alternative source for me? thx