Qengineering/Jetson-Nano-Ubuntu-20-image #46: xrdp remote access

kyrie2to11, 2023-08-31:
Thank you for the image! 
After flashing and installing xrdp, I connect to my jetson nano B01 and the screen freezes as shown below. 
[image]
Do you have some hints that can help with this problem please?

Qengineering, 2023-08-31:
@kyrie2to11,

I'm sorry to hear about your issue. More over, because I think there is not a quick fix.
I've googled 'xrdp jetson nano crash' and got 38.300 hits. 
Looking at the first ones, it seems a common issue with a variety of solutions. 
In your cast, I would suggest to use Teamviewer. see also https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/43.

kyrie2to11, 2023-08-31:
Thank you for your reply. I will attempt to use Teamviewer.
Additionally, I have conducted the xrdp connection testing with Jetpack 4.6.1 and it functions flawlessly with almost no delay.

drjajas, 2023-09-01:
@kyrie2to11 so for the following did the trick: 

sudo apt install xserver-xorg-video-dummy

sudo nano /usr/share/X11/xorg.conf.d/xorg.conf

Input the following in the file:

```
Section "Device"
Identifier "Configured Video Device"
Driver "dummy"
EndSection

Section "Monitor"
Identifier "Configured Monitor"
HorizSync 31.5-48.5
VertRefresh 50-70
EndSection

Section "Screen"
Identifier "Default Screen"
Monitor "Configured Monitor"
Device "Configured Video Device"
DefaultDepth 24
SubSection "Display"
Depth 24
Modes "1920x1080"
EndSubSection
EndSection
```

After that reboot.

drjajas, 2023-09-07:
i found some other ways:

sudo nano /etc/xrdp/startwm.sh

in the file comment out the last two lines:
`
#test -x /etc/X11/Xsession && exec /etc/X11/Xsession
#exec /bin/sh /etc/X11/Xsession
`
and add one of the following at the end:
`
1. 
gnome-session
2. 
startlxde
3. 
startxfce4
you need to install this first if you want to use it:
sudo apt install xfce4 xfce4-terminal -y
`

after that you need to restart xrdp:
`sudo systemctl restart xrdp.service`