Qengineering/Jetson-Nano-Ubuntu-20-image #48: Missing linux-headers-4.9.253

tganderson0, 2023-10-10:
Hello, I was trying to install https://github.com/WCHSoftGroup/ch343ser_linux and hit some issues when running `make`. I believe this is due to the bare image, as the error I get is:
`make[1]: *** /lib/modules/4.9.253/build: No such file or directory.`

From what I could tell, it seems the linux-headers-4.9.253 are missing, which is the output of `uname -r`, but when I try installing headers, all I can get are 5.x. Wondering if anyone has solutions for this!

Qengineering, 2023-10-11:
That's weird. It would mean that the software only works with one specific JetPack (4.9.253). I can't imagine that.

Loading an extra kernel is risky. In fact, you have two OSes on your system. This easily leads to conflicts.

That's why I tried to install the package myself.
And, as it turns out, it works straight away. Without any problems.

[image]

I don't know what is causing your error, but it most likely won't be the WCHSoftGroup software.

tganderson0, 2023-10-17:
Ended up going back to the full image that included OpenCV instead of the base one, and it worked just fine. It might be something weird with the base image, but since it worked fine with the full image I'm happy with it!

closed 2023-10-17 by tganderson0