Qengineering/Jetson-Nano-Ubuntu-20-image #49: whereis NVCC tha I can't find & operate with?

wir9, 2023-10-11:
I see /usr/local/cuda-10.2, but "nvcc --version" has no response. Thanx in advance. But is very useful, that I can finally use Torch, my target. Thanx a lot. I wish to use .cu to compile in vscode,

benjmerales, 2023-10-23:
Hi, let me help. Does using `nvcc -V` also not work?

Qengineering, 2023-10-23:
Must work.
[image]

closed 2023-10-23 by Qengineering

wir9, 2023-10-23:
Not work.
One more thing.
Installed Intel wifi card, pasted iwil.deb to lib/firmware and keep not
working.
What I am doing badly?.
Thanx un advance

Qengineering, 2023-10-23:
Sorry, that is the problem with many issues we received.
Everything works when deliverer. People start installing there software and get stocked.
Understandable, but (almost) impossible to debug.
I have really no clue. Something must have meshed with the Ubuntu/ NVIDIA folder structure.

wir9, 2023-10-23:
Yes, that's I think, too
I was going to acquire a new Jetson orin nano, but due to the fails of this
Jetson nano I will not. I've seen that Intel 9461 is alright, but, why not
the rest???.
Thanx for your time and attention.
Have a nice day!.
nvcc -V only goes when root, not user $.
How should I solve this.
It implies not recognized in VA code Studio.
Sorry.