Qengineering/Jetson-Nano-Ubuntu-20-image #50: Flashing on 128GB SD doesn't use its full storage

benjmerales, 2023-10-19:
I just recently flashed my SanDisk Extreme microSDXC memory card with the image. It works, but it only uses up to 32GB of space.
SanDisk Extreme microSDXC memory card.

[image]

Qengineering, 2023-10-19:
@benjmerales,

That is common behaviour when flashing a large SD card with an image originally smaller.

You can enlarge the partition (in your screenshot **/dev/mmcblk0p1**) by using a tool called [GParted](https://gparted.org/) 

`sudo apt-get install gparted`

benjmerales, 2023-10-19:
Nice! This worked flawlessly!

Thank you

gpullela, 2023-11-13:
Hello - I am getting the following error when I run gparted.  
[image]
I ran the ```sudo apt-get install gparted``` command to install gparted but it is not letting me resize the partition on my sd card

Qengineering, 2023-11-13:
You need to use another PC to do the trick. It is not possible the resize any partition on the SD-card you are currently using.