Qengineering/Jetson-Nano-Ubuntu-20-image #52: Upgrading Tensorflow

benjmerales, 2023-10-23:
I have a saved H5 model that is built on Tensorflow 2.12. I successfully converted it to a compatible TensorRT version. But it seems there are some layers that 2.4 is yet to discover (Keras's _keepdims_ keyword). 

Is there right way to install the latest version of Tensorflow? One that can determine the Nano's GPU device. Since normally using `pip install -U tensorflow` will cause TF to restrict the GPU.

Qengineering, 2023-10-23:
@benjmerales,

`pip install` will not give you CUDA acceleration. 

I'm afraid, it's even worse.

Jetson Nano can't support TensorFlow > 2.4.1. with CUDA, as these TF versions need CUDA 11, which can't be installed on the nano due to low-level incompatibility.

benjmerales, 2023-10-23:
Bummer. Guess my only solution now is to create a new model that can support TensorFlow 2.4.1, huh.

Thanks. Closing this issue now.

closed 2023-10-23 by benjmerales