Qengineering/Jetson-Nano-Ubuntu-20-image #54: Updating kernel from 4.9.253?

akielaries, 2023-10-30:
I am using your base 20.04 image and notice the kernel was on an old version still. Is it possible to upgrade the kernel on your image or will this break some existing compatibility? 

I am trying to install the CUDA toolchain but had seen the warning in the `README` that v11 and beyond are not compatible with the Jetson's hardware, what version and kernel driver should we target for this image?

Qengineering, 2023-10-30:
Updating the kernel is very risk full. You can easily end up with conflicting dependencies.
Perhaps in the end, you may succeed. However I would strongly recommend not to do so.
Besides, the CUDA tool chain is already installed as you can see below.
[image]

akielaries, 2023-10-30:
Ah interesting, I had done a fresh install of the base image but noticed none of the CUDA toolkit was included in image. Is the base CUDA toolchain (not TensorFlow or OpenCV related libs) not included in the base installation?

akielaries, 2023-10-30:
Let me try a fresh installation once more and list what dpkg is aware of

akielaries, 2023-10-30:
When I was trying to do my own installation of a more recent version of CUDA and its driver it had bricked the initial installation from the image. Reinstall worked and was able to have the toolchain all present

```
jetson@nano:~$ dpkg -l | grep cuda
ii  cuda-command-line-tools-10-2               10.2.460-1                                 arm64        CUDA command-line tools
ii  cuda-compiler-10-2                         10.2.460-1                                 arm64        CUDA compiler
ii  cuda-cudart-10-2                           10.2.300-1                                 arm64        CUDA Runtime native Libraries
ii  cuda-cudart-dev-10-2                       10.2.300-1                                 arm64        CUDA Runtime native dev links, headers
ii  cuda-cuobjdump-10-2                        10.2.300-1                                 arm64        CUDA cuobjdump
ii  cuda-cupti-10-2                            10.2.300-1                                 arm64        CUDA profiling tools runtime libs.
ii  cuda-cupti-dev-10-2                        10.2.300-1                                 arm64        CUDA profiling tools interface.
ii  cuda-documentation-10-2                    10.2.300-1                                 arm64        CUDA documentation
ii  cuda-driver-dev-10-2                       10.2.300-1                                 arm64        CUDA Driver native dev stub library
ii  cuda-gdb-10-2                              10.2.300-1                                 arm64        CUDA-GDB
ii  cuda-libraries-10-2                        10.2.460-1                                 arm64        CUDA Libraries 10.2 meta-package
ii  cuda-libraries-dev-10-2                    10.2.460-1                                 arm64        CUDA Libraries 10.2 development meta-package
ii  cuda-memcheck-10-2                         10.2.300-1                                 arm64        CUDA-MEMCHECK
ii  cuda-nvcc-10-2                             10.2.300-1                                 arm64        CUDA nvcc
ii  cuda-nvdisasm-10-2                         10.2.300-1                                 arm64        CUDA disassembler
ii  cuda-nvgraph-10-2                          10.2.300-1                                 arm64        NVGRAPH native runtime libraries
ii  cuda-nvgraph-dev-10-2                      10.2.300-1                                 arm64        NVGRAPH native dev links, headers
ii  cuda-nvml-dev-10-2                         10.2.300-1                                 arm64        NVML native dev links, headers
ii  cuda-nvprof-10-2                           10.2.300-1                                 arm64        CUDA Profiler tools
ii  cuda-nvprune-10-2                          10.2.300-1                                 arm64        CUDA nvprune
ii  cuda-nvrtc-10-2                            10.2.300-1                                 arm64        NVRTC native runtime libraries
ii  cuda-nvrtc-dev-10-2                        10.2.300-1                                 arm64        NVRTC native dev links, headers
ii  cuda-nvtx-10-2                             10.2.300-1                                 arm64        NVIDIA Tools Extension
ii  cuda-repo-l4t-10-2-local                   10.2.460-1                                 arm64        cuda repository configuration files
ii  cuda-samples-10-2                          10.2.300-1                                 arm64        CUDA example applications
ii  cuda-toolkit-10-2                          10.2.460-1                                 arm64        CUDA Toolkit 10.2 meta-package
ii  cuda-tools-10-2                            10.2.460-1                                 arm64        CUDA Tools meta-package
ii  cuda-visual-tools-10-2                     10.2.460-1                                 arm64        CUDA visual tools
ii  graphsurgeon-tf                            8.2.1-1+cuda10.2                           arm64        GraphSurgeon for TensorRT package
ii  libcudnn8                                  8.2.1.32-1+cuda10.2                        arm64        cuDNN runtime libraries
ii  libcudnn8-dev                              8.2.1.32-1+cuda10.2                        arm64        cuDNN development libraries and headers
ii  libcudnn8-samples                          8.2.1.32-1+cuda10.2                        arm64        cuDNN documents and samples
ii  libnvinfer-bin                             8.2.1-1+cuda10.2                           arm64        TensorRT binaries
ii  libnvinfer-dev                             8.2.1-1+cuda10.2                           arm64        TensorRT development libraries and headers
ii  libnvinfer-doc                             8.2.1-1+cuda10.2                           all          TensorRT documentation
ii  libnvinfer-plugin-dev                      8.2.1-1+cuda10.2                           arm64        TensorRT plugin libraries
ii  libnvinfer-plugin8                         8.2.1-1+cuda10.2                           arm64        TensorRT plugin libraries
ii  libnvinfer-samples                         8.2.1-1+cuda10.2                           all          TensorRT samples
ii  libnvinfer8                                8.2.1-1+cuda10.2                           arm64        TensorRT runtime libraries
ii  libnvonnxparsers-dev                       8.2.1-1+cuda10.2                           arm64        TensorRT ONNX libraries
[... cut ...]
ii  nvidia-container-csv-tensorrt              8.2.1.8-1+cuda10.2                         arm64        Jetpack TensorRT CSV file
ii  nvidia-l4t-cuda                            32.7.2-20220420143418                      arm64        NVIDIA CUDA Package
ii  tensorrt                                   8.2.1.8-1+cuda10.2                         arm64        Meta package of TensorRT
ii  uff-converter-tf                           8.2.1-1+cuda10.2                           arm64        UFF converter for TensorRT package
```

closed 2023-10-30 by akielaries