Qengineering/Jetson-Nano-Ubuntu-20-image #59: A Jetson Nano Ubuntu 20.04 with  OpenCV, TensorFlow and Pytorch BUT OVERCLOCKED?

srevandro, 2023-11-15:
Dear @Qengineering,

How are you? 

Thanks for Jetson Nano Ubuntu 20.04 image. I would like to know with you don´t have the Ubuntu 20.04 with OpenCV, Tensorflow, Pytorch etc, but overclocked?  I just saw here the Bare Image (without any ML library) that is overclocked. 

I have tried to your tutorial to overclock the Ubuntu 20.04 with all libraries, but did not succeed. I just get some errors:

make[2]: *** [scripts/Makefile.build:649: drivers/net/wireless] Error 2
make[2]: *** Waiting for unfinished jobs....
  CC [M]  drivers/net/veth.o
  CC [M]  drivers/net/macvlan.o
  CC [M]  drivers/net/macvtap.o
make[1]: *** [scripts/Makefile.build:649: drivers/net] Error 2
make: *** [Makefile:1122: drivers] Error 2

I have looked at the NVIDIA forum, but did not succeed as well. 
https://forums.developer.nvidia.com/t/overclocking-jetson-nanos-cpu-and-gpu/83501/11

Do you have any idea how to solve this problem? I have tried to copy, but did not work.
sudo cp /proc/config.gz $TEGRA_KERNEL_OUT/

Another possibility is do you have any tutorial about the installation of OpenCV, Tensorflow, Pytorch in the Bare Image? I think should be easier.

Thanks.

Qengineering, 2023-11-15:
I've chosen not to overclock the Nano because some frameworks may become unstable. Not that it will happen, but chances are. 
Other frameworks don't profit from overclocking due to their use of CUDA.
The second reason is that the files involved in overclocking are frequently updated by NVIDIA. After a "sudo apt-get upgrade" or the software updater, overclocking can be undone.
On the other hand, apart from these remarks, there is no reason overclocking won't work.
The procedure is always the same. Get a fresh Tegra OS, overclock and then upgrade to Ubuntu 20.
If you want to install frameworks on the bare (and overclocked) image, please use our guides on the website. 
We also use them to compose this SD image.

srevandro, 2023-11-17:
Dear @Qengineering ,

I have done some attemps to get Jetson nano overclock. Following I will list my procedures and problems that I get:

**1) I have got you Ubuntu 20.04 LTS FULL (with Pytorch, OpenCV etc), and I tried to overclock it. It does not work. I got the previous errors mentioned before.**
`make[2]: *** [scripts/Makefile.build:649: drivers/net/wireless] Error 2
make[2]: *** Waiting for unfinished jobs....
CC [M] drivers/net/veth.o
CC [M] drivers/net/macvlan.o
CC [M] drivers/net/macvtap.o
make[1]: *** [scripts/Makefile.build:649: drivers/net] Error 2
make: *** [Makefile:1122: drivers] Error 2`

**2) I get the original sd card image from NVIDIA with Ubuntu 18.04 LTS** (https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write), and tried to overclock it. On this case I got now drivers missings errors and succeed on the overclock. However the big problem in this version is that they use Python 3.6.9, and some libraries (such as matplotlib pillow etc) requires at this Python 3.7.

**3) I get you Bare Image already overclocked.** and following your tutorials on your site I succeed in install Pytorch. However it seems GPU(CUDA) is not working on this case. **_So my first question if I ONLY need to do the procedures in the "Installation by wheel" or should I do any other to get GPU(CUDA) working?_** 

4) Do you think I also will have problems with GPU (CUDA) in the installation of torchvision, OpenCV, Tensorflow ?

5) Last question, is a silly one, but I need to be sure I am doing the correct thing. Suppose I install you Bare Image with overclocked system, and later on I install Jetson-Inference container and Jetson-Containers, and try to run the libraries torch, torchvision etc from this containers. My question if this containers will use the CPU (1.9GHz) and GPU (0.95GHz)  overclocked or not since they are containers?
 
Thanks for all you support, and I will be looking forward your support. It´s really important to me succeed on this issue, because I really need an overclocked system with all deep learning libraries install.

Sincerely yours,
Evandro

Qengineering, 2023-11-17:
1) It is possible the install OpenCV, TensorFlow and PyTorch on the bare overclocked image. I've tested it with PyTorch. 
Below the result.

[image]

2) I'm uploading the SD-card to Sync. I will share the link as soon as the upload is completed. It holds PyTorch. Not OpenCV nor TensorFlow. If you don't mind, you can give it a try.

3) At the moment you upgrade your system (or your clients) you will lost the overclocking. Don't use `sudo apt-get upgrade`.

4) Not sure if the Jetson Containers are holding the complete OS. (Seems to be??)

srevandro, 2023-11-17:
Dear @Qengineering , 

I am really really grateful for all you help. 

I was a little bit confused with NVIDIA instructions to update the pytorch version, since they recommend to update a version where it was using Python3.6. About the CUDA is working perfectly. This was my own mistake.

For all day long I tried the Bare image, but I have done "sudo apt update && sudo apt upgrade". Even after the update and upgrade when I execute the "sudo /usr/bin/jetson_clocks --show", I still getting MaxFreq=1900000.  **Only this information is enough to guarantee that the Jetson is overclocked?** 

If I try to get your Ubuntu 20.04 with Pytorch, OpenCV, Tensorflow etc, and apply just the steps described on you page, with the Title "Max frequency", and use the overclock.sh script, is that enough to get the jetson nano overcloked or I really need to rebuild the whole kernel?

Dear friend, I am writing an article about jetson nano inferences. I would like to add my thanks to you for all the help and the tutorial on the site too, how can I reference it? Just mention to the author of the Qengineering website as reference?

Thanks.

Qengineering, 2023-11-18:
1) `sudo apt-get upgrade` isn't the problem. The problem arise as soon as the upgrade procedure wants to (partly) replays the kernel. That's were things go wrong. A regular upgrade of FireFox, for instance, is no problem.

2) The `/usr/bin/jetson_clocks --show` reflects the values in the governor, the values wherein the clock may vary. If it says 1900, you know your Jetson Nano is still overclocked.

3) Please find an already overclocked Ubuntu 20.04 version with PyTorch. The one giving the screen dump from above. No need to rebuild the kernel. That's done. You only have to install OpenCV en TensorFlow yourself. The Image is about 7 GB.
https://ln5.sync.com/dl/778d47210/pkg4aht5-7mt4hxb3-k4t2pc4g-zz6rxjwh

4) Reference to the website would be appreciated.

srevandro, 2023-11-18:
Thanks my friend. You are the best. Regards.

closed 2023-11-18 by srevandro