Qengineering/Jetson-Nano-Ubuntu-20-image #62: Load image Error

AeolianLbells, 2023-12-11:
I followed the instructions in the documentation and burned the image of this project, JetsonNanoUb20_3b.img.xz, using Pi imager to the SD card, which is 64G in size. I took the SD card out of the USB card reader and put it into the jetson nano, then powered it on and booted it. The screen showed the Nividia logo and then executed the content in the first picture, and then it stopped at the content in the second picture and did not move. Has anyone encountered the same problem as me?

my device is jetson nano 4G B01.
[image]
[image]

AeolianLbells, 2023-12-11:
i think its the same problem with emmc

closed 2023-12-12 by AeolianLbells

Qengineering, 2023-12-12:
How did you solve it?
I could not reproduce the issue at the office.

krizek00, 2023-12-12:
I encountered practically identical errors after flashing JetsonNanoUb20_3b.img.xz using Pi Imager on a 128 GB memory card. The device I am using is the reComputer J1010 - Nvidia Jetson Nano 4GB RAM. Today, I tried to flash an older version, JetsonNanoUb20_2.img, but this time different errors appeared. I read on the NVIDIA forum that it is not possible to install Ubuntu 20.04 on this board. Is this really the case? Regards.
[image]
[image]

Qengineering, 2023-12-12:
Do you have the link to the forum topic?

As far as I can see, it seems it might have to do with hardware differences. 
In my case everything works flawless. 
Perhaps we must do a custom upgrade from the original JetPack to Ubuntu 20.04, according our [guide](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html).

krizek00, 2023-12-12:
https://forums.developer.nvidia.com/t/jetson-j1010-flasing-issue/249200/2

Thanks for the response. This thread doesn't directly address the issue because the installation was done through the SDK Manager, but it is related to J1010. The only thing is that the NVIDIA moderator mentioned that flashing Ubuntu 20.4 on Jetson Nano is not possible, although it is achievable through your solution :). I'll try upgrading JetPack using the link you provided.