Qengineering/Jetson-Nano-Ubuntu-20-image #63: How to use Arducam IMX519 in this Version

xmod3905, 2023-12-15:
I have problem with this version while installing Arducam IMX519 also know as Arducam 16mp. It is possible to implement it in this version?
[image]

Qengineering, 2023-12-15:
Not sure....
Does it work on a regular Jetson Nano with Ubuntu 18 (Original Tegra jetpack)?
Whe it worked on that OS, it should work also on the Ubuntu 20.04 version.

xmod3905, 2023-12-15:
That's works with Ubuntu 18, but with default Jetpack Version

xmod3905, 2023-12-18:
Thare any solution to solve this problem?

Qengineering, 2023-12-18:
At this very moment, I'm uploading a new Ubuntu 20.04 to the cloud.
As soon as it is completed, I will let you know. However, it is a large file. It will take a while.

It may not solve your issue. IMX519 requires a JetPack 32.6.1 or higher. 
The current Ubuntu 20.04 is built upon 32.6.1. It should work (??).
However, to be sure, the current upload uses JetPack 32.7.4. It has no additional software, just a bare version to test. 

The second possible cause can be the GNU version. Tegra has C++ 8, while Ubuntu ships with version 9.
Please look at our website](https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html) on how to switch C++ versions.
If it still doesn't work, we have a problem. I have no insides on how ArduCam integrates on a low-level driver base with the Jetson Nano.

Qengineering, 2023-12-21:
It took some time, but here it is: https://ln5.sync.com/dl/a3db65ab0/dtwyjs53-gbep3jqf-npi66vgc-i9c4s2aa 
Ubuntu 20.04 based on JetPack 32.7.4.

SentryCoderDev, 2024-01-21:
Unfortunately it didn't happen again, it still gets canceled halfway through.

xmod3905, 2024-03-27:
sorry i'm so late but thanks men