Qengineering/Jetson-Nano-Ubuntu-20-image #64: Compile OpenCV on new images with GStreamer and CUDA support

GNCGenie, 2023-12-22:
Just a QOL improvement, if for future releases OpenCV could be built against the installed GStreamer libraries and with CUDA support it would make this image perfect!

Qengineering, 2023-12-22:
I'm using [this script](https://github.com/Qengineering/Install-OpenCV-Jetson-Nano/edit/main/OpenCV-4-8-0.sh) to install OpenCV. It supports the `-D WITH_GSTREAMER=ON` flag. 
I tough it was sufficient to get GStreamer into OpenCV. Assuming GStreamer has its CUDA acceleration up and running.
If not, how would you do it?
BTW, most users like to build their pipelines with the nv* NVIDIA modules, like `nvv4l2decoder` or `nvvidconv`

GNCGenie, 2023-12-23:
Yes, I just checked the build info on a fresh board, and it is built with GStreamer enabled. 
I must have altered it accidentally!

closed 2023-12-23 by GNCGenie