Qengineering/Jetson-Nano-Ubuntu-20-image #66: Link Error

SentryCoderDev, 2024-01-21:
Hello, I love your project. I'm trying to download it because I can't run jetson-io on my own Jetson, but I have a problem. I tried to download it again 7 times and it always gets canceled halfway through(5 diffrent device(Bare Image)). The same thing happened to the others and I haven't been able to download it for 4 days. Can you upload the Bare Image version to google drive or a similar cloud and send the link? ? Jetson-io program is very important for me. It gives an error about curses. Maybe it will work with your OS.

Qengineering, 2024-01-21:
Look at th end of [this issue](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/63).
You will find a link. Please try it. I'm not sure it will help you, because it is still one large download.
Let me know your findings. Otherwise I will provide a split image.
P.S. Google Drive only permits a few downloads a day. This image is very popular, and has over 300 downloads a day.
That's why we don't use Gdrive.

SentryCoderDev, 2024-01-21:
Unfortunately it didn't happen again, it still gets canceled halfway through.

SentryCoderDev, 2024-01-22:
Could you please upload the file to drive you provided (link in the issues) and send it to me, not for everyone to use but just for once?

I tried each file one by one. The download speed is reset for every 100 MB download. When I download 200 or 300 MB, the file is canceled.

I need to try its OS urgently, maybe I might sell my Jetson.

Qengineering, 2024-01-22:
I've uploaded the  latest bare image (with OpenCV) to Google Drive.
[Here is the link.](https://drive.google.com/file/d/1yNPl9LDvQiB4euu8Hl1UkaBDQ9SA5LuJ/view?usp=drive_link)