Qengineering/Jetson-Nano-Ubuntu-20-image #67: jetson-io.py issue

SentryCoderDev, 2024-01-22:
Hello, I installed your OS, but unfortunately I still have the same problem. When I run the 40 pin expansion header (jetson-io.py), it turns on and off for half a second.

Do you have a solution suggestion?

Qengineering, 2024-01-22:
I'm sorry. I'm not familiar with jetson-io.py. 
Given the number of posts on the NVIDIA forum, it looks there are a lot of similar issues.
And all pretty low-level. Also with there solutions.
One thing for sure, it is not your hardware.

SentryCoderDev, 2024-01-23:
Hello, after reinstalling it was fixed. Your OS will be very useful to me, thank you again.

closed 2024-01-23 by SentryCoderDev

nick-pape, 2024-03-25:
I also am getting this issue, with the error stating that no APP partition was found (when it runs `