Qengineering/Jetson-Nano-Ubuntu-20-image #69: Download speed is horrible

pbrandstet, 2024-01-24:
The download speed for me is ~100kB/s
Maybe you can upload the file on a stronger server.

JustZeus, 2024-01-26:
For me the speed was around 7 MB/s and at 3.4 GB it stopped downloading, three times tried and same result.

Hope you can add an extra link in another server as @pbrandstet mentioned.

Thanks!

Qengineering, 2024-01-26:
Nothing is more annoying than a slow or broken connection. We know.
That's why we provide a split image: instead of the huge 8.7 GB, you can download smaller portions. They are merged during unzipping. 
Google Gdrive only allow a limited amount of downloads per day. Once reached, you have to wait 24 hours before Gdrive re-opens. Mega and others limit the file size. Sync is tested one of the best. That's why we use this server for now.

pbrandstet, 2024-01-26:
I tried the splitted image, but with the splitted image i get an nginx error (i think 503)

kpma1985, 2024-01-29:
[image]

pbrandstet, 2024-01-29:
Now its possible to download the splitted image.
But it always only loads 10MB, then it stops for a while an downloads another 10MB. Or it just losses the connection and stop the download.

nobrowser, 2024-08-26:
Hello.

I really appreciate your work on this, and I'd like to try it. But as it is, I can not. I will only use the images if you put them behind real, transparent links. There are (at least) two problems with the way you do it now:

- I have to babysit my browser for hours while it works on the downloads, instead of spawning a curl job in a screen session and forgetting about it, which is the way I normally do large downloads
- I can't shake a lingering suspicion that this is some kind of scam to take advantage of people like me, desperate to make some use of our Jetsons after Nvidia effectively "doorstopped" them.

I look forward to your reply, and thank you.

-- 
Ian

thanhtruongphan, 2024-11-05:
@pbrandstet I have same with you... The download speed  ~100kb/s 

do you have this for your own, let share with me, pls :smiley_cat: 

-thank you-

Qengineering, 2024-11-05:
In some parts of the world it is difficult to get a good solid connection to Sync. That is why we've also provided a copy on [Google Drive](https://drive.google.com/file/d/1L2H_sQC_kSILrcJteWg7htKxJirtDsZ9/view?usp=sharing). However, Google Drive limits the number of daily downloads, which is much lower than our average daily download volume. Please be considerate and only use Google Drive if necessary.

thanhtruongphan, 2024-11-05:
great!~  @Qengineering 

Nam Mô A Di Đà Phật... 

it's about ten minutes ... 
Thank you so much! 

:100: