Qengineering/Jetson-Nano-Ubuntu-20-image #71: Problems with VNC

olegg366, 2024-02-14:
There is a problem with connecting to jetson nano 4gb via vnc. I used the official [guide](https://developer.nvidia.com/embedded/learn/tutorials/vnc-setup). It worked with yours ubuntu 18.04, but with this image, using RealVNC, I cant connect to my jetson. It just says "time out wating for responce from remote computer". Can someone help?

Qengineering, 2024-02-14:
We will address the issue asap.

olegg366, 2024-02-14:
Thanks! I'll be looking forward to it.

olegg366, 2024-02-24:
Thanks, but I don't have a monitor to follow the steps you indicated. Could you translate these instructions into terminal commands? I think they differ from nvidia docs only in vnc client settings. But even with this step, I can't connect to jetson.

doodieb15, 2024-02-24:
Any chance to enable VNC on headless Jetson Nano?

Qengineering, 2024-02-24:
I'm working on a headless solution. I need to overwrite the default screen resolution. The well known Xorg.conf doesn't work.

Borja21091, 2024-03-05:
I managed to connect following the steps from the README, but after the first connection it looks like the live feed is not working any more. The date&time in the Jetson's Desktop shows the time when I first connected to it, although it's the following day now.

To be clear, I can see the mouse moving on my VNC Client, but clicking and text input does not work. Anyone has experienced the same behaviour? Definitely weird.