Qengineering/Jetson-Nano-Ubuntu-20-image #82: Error when importing torchaudio

MarkHmnv, 2024-04-21:
Hello, after installing the image I installed torchaudio
```bash
pip3 install torchaudio==0.13.0
```
But when I try to import it I get the following error:
```bash
Traceback (most recent call last):
  File "inference_record.py", line 8, in <module>
    import torchaudio
  File "/home/jetson/.local/lib/python3.8/site-packages/torchaudio/__init__.py", line 1, in <module>
    from torchaudio import (  # noqa: F401
  File "/home/jetson/.local/lib/python3.8/site-packages/torchaudio/_extension.py", line 135, in <module>
    _init_extension()
  File "/home/jetson/.local/lib/python3.8/site-packages/torchaudio/_extension.py", line 105, in _init_extension
    _load_lib("libtorchaudio")
  File "/home/jetson/.local/lib/python3.8/site-packages/torchaudio/_extension.py", line 52, in _load_lib
    torch.ops.load_library(path)
  File "/usr/local/lib/python3.8/dist-packages/torch/_ops.py", line 573, in load_library
    ctypes.CDLL(path)
  File "/usr/lib/python3.8/ctypes/__init__.py", line 373, in init
    self._handle = _dlopen(self._name, mode)
OSError: /home/jetson/.local/lib/python3.8/site-packages/torchaudio/lib/libtorchaudio.so: undefined symbol: _ZNK5torch8autograd4Node4nameEv
```
This error did not appear on the original jetpack image

Qengineering, 2024-04-21:
Please see: https://forums.developer.nvidia.com/t/problems-with-torchaudio-install-in-ubuntu-20-and-python38/204715

MarkHmnv, 2024-04-21:
> Please see: https://forums.developer.nvidia.com/t/problems-with-torchaudio-install-in-ubuntu-20-and-python38/204715

Yes, I have attached that link in my post with the problem. I understand I need to build torchaudio manually?

Qengineering, 2024-04-21:
I see. Sorry, missed the link.
I seems as we have some mismatch between versions.
The solutions suggest a manual rebuild.
You have probably seen this also: https://github.com/pytorch/audio/issues/62

MarkHmnv, 2024-04-30:
Building torchaudio manually worked

closed 2024-04-30 by MarkHmnv