Qengineering/Jetson-Nano-Ubuntu-20-image #83: Usb boot

21coolkarni, 2024-04-28:
Can I use this os with usb boot on jetson nano

Qengineering, 2024-04-30:
I've never tried it. Be my guest and let us all know if you have success.

Pratik261201, 2025-06-12:
Hi! 👋 I ran into a similar issue while working with the Jetson Nano on a Waveshare carrier board, particularly when trying to boot from an SD card using a custom image flashed via Windows.

After some troubleshooting, I was able to resolve the problem and have documented the complete working solution here:

🔧 Solution Repo: https://github.com/Pratik261201/Jetson-Nano-Support

Here’s a quick overview of what worked:
 configure the boot order correctly to ensure the SD card is prioritized.

You’ll find step-by-step instructions in the repository, organized into folders with individual README.md files containing detailed guidance for each part of the process.

Feel free to check it out — happy to help if you run into any further issues!