Qengineering/Jetson-Nano-Ubuntu-20-image #84: Flashing cursor - no GUI

pete-789, 2024-04-30:
First of all thank you so much for creating this image, getting 20.04 (and thus Python 3.8) to run on these machines is really important to a lot of people.  My unit is a Seeed J1020 - basically a 4GB Nano with a 16GB eMMC and a NVME slot underneath.

I've got the original (18.04) Jetson image on the 16GB eMMC, and this 20.04 image on a 256GB SSD drive in the NVME slot. I used the technique mentioned in Issue #32 and modified the extlinux.conf file on both devices. For my setup '/dev/nvme0n1p1' works.

At boot, after a few lines on the screen, the screen clears and there is just a flashing cursor in the top corner. No GUI.

I can log in to the Jetson-Nano-Ubuntu-20-image okay using the serial console, so it is booting okay.

Is this image intended to have a GUI, or is it 'headless'?  I'd like a GUI, what can I do - or what could be wrong?

Many thanks for any help you can give!

Qengineering, 2024-04-30:
I can't help you with this issue, as I don't have your configuration.
The image is intended to have a GUI. That's for sure.
Some of the people of issue #32 might help you.

pete-789, 2024-05-01:
I think this might be an HDMI / DVI issue - at boot I get 

`tegra-i2c 7000c000.i2c: no acknowledge from address 0x50 `
and 
`tegra-i2c 7000c400.i2c: no acknowledge from address 0x50`

If I boot without connecting the HDMI cable, then connect it I get

`edid invalid`

I started a [thread on the NVIDIA forum] (https://forums.developer.nvidia.com/t/boots-to-flashing-cursor-what-next/291163/7) and they thought that the 'device tree' might need editing? Or might there be something else that is included in the original 18.04 build (the monitor works fine) but missed off the QEngineering 20.04 build?  Something I could install or edit maybe?

(I will also try to find a true HDMI monitor to test with).

I'm excited because this seems so close to working!

DaveBlutix, 2024-12-09:
I have the same configuration, have you managed to make it work?

vggscqq, 2025-06-25:
> I am using the HDMI port, through a DVI

As per, 
https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#troubleshooting
HDMI to DVI adaptors are not supported. Please use a display that accepts HDMI or DP input.

fanel22, 2025-09-01:
> First of all thank you so much for creating this image, getting 20.04 (and thus Python 3.8) to run on these machines is really important to a lot of people. My unit is a Seeed J1020 - basically a [...]

Did just modifying with nvme0n1p1 work or did you change with dev/partuuid = ....  ? I have same board but doesn't work to change the boot how it should be

fanel22, 2025-09-02:
If i try to install Bare Image will work this software on J1020?