Qengineering/Jetson-Nano-Ubuntu-20-image #85: Tensorflow GPU support not working

Nicholas-B1, 2024-05-03:
Hello, I installed this OS for my Jetson Nano but I cannot get Tensorflow to recognize my GPUs being installed on the board.

Qengineering, 2024-05-03:
Please note that only TensorFlow 2.4.1 or a lower version will work with the Jetson Nano.
Above requires CUDA 11, which is not available for the Jetson Nano.

Nicholas-B1, 2024-05-03:
I have tensorflow 2.13. Is there anything else I need for configuring GPU support with this flashed OS?

Qengineering, 2024-05-03:
You cannot run TF 2.13 on a Jetson Nano other than in the CPU mode. If you want GPU, you need to downgrade TF to version 2.4 or lower.

Nicholas-B1, 2024-05-03:
I cannot find version of tensorflow that old with the python version that comes installed on this flashed OS.

Qengineering, 2024-05-04:
When you look at: https://www.tensorflow.org/install/source,
you see these lines:
[image]
There you go.

Nicholas-B1, 2024-05-17:
Okay thank you I can try that. So the flashed image should be able to recognize the jetson nanos GPU right away or is there any configuration needed?

Qengineering, 2024-05-17:
It recognize the CUDA cores out of the box. There is no need for additional configurations.