Qengineering/Jetson-Nano-Ubuntu-20-image #87: Boot Problem

kianNaderi, 2024-05-09:
[image]
Hi, I downloaded the provided image and flashed it onto an SD card using Etcher. I have a Waveshare carrier board with a Jetson Nano, so I followed Waveshare's instructions to ensure that the system recognized the SD slot. After that, I modified the extlinux.conf file to boot from the SD card slot where I had flashed your image (/dev/mmcblk1p1). However, when I rebooted, I encountered several errors. I used windows for flashing images. I asked Waveshare support but thought asking here could also help. Thanks for your attention

FlorianShepherd, 2026-01-05:
Hi @Pratik261201 your link gives me a 404 - I'd appreciate a lot if you could share the guide again :)

FlorianShepherd, 2026-01-05:
You're the best. That works! Thank you @Pratik261201 !