Qengineering/Jetson-Nano-Ubuntu-20-image #88: Can we get an Headless image?

KalanaRatnayake, 2024-05-15:
Hey all!

Thanks for your hard-work on Jetson Nano related stuff.

I have started working with your bare image for setting up an Jetson Nano we have at the lab for robotic applications (hoping to run ROS2 applications). I was setting it up and when looking at htop/jtop it shows around a usage between 1GB - 2GB RAM usage.
 
[image]

Above image is an newly setup nano today and it has a usage of about 1.5GB usage.

I tired to go headless (As I heard GNOME takes about 800MB), disabling the desktop and temporarily and permanently (had to re-flash the SD card after this) but didn't manage to reach the tty terminal. Wanted to try to free up more usable ram (I usually do Ubuntu server on Raspberry pi)

Are there Ubuntu server images pre-setup like this for jetson nano? Sorry if I am asking to drive a car without wheels.

Qengineering, 2024-05-15:
Here are the instructions.
```
sudo chown root:root / /lib
sudo apt purge ubuntu-desktop -y && sudo apt autoremove -y && sudo apt autoclean
sudo apt-get remove nautilus nautilus-* gnome-power-manager gnome-screensaver gnome-termina* gnome-pane* 
sudo apt-get remove gnome-applet* gnome-bluetooth gnome-desktop* gnome-sessio* gnome-user* gnome-shell-common
sudo apt-get remove zeitgeist-core libzeitgeist* gnome-control-center gnome-screenshot && sudo apt-get autoremove
sudo apt-get remove --purge libreoffice*
sudo apt-get remove libreoffice-core
sudo apt-get remove snapd lightdm cups chromium*
```
You end up with a headless OS consuming ~ 420 MB.
Below is a screen dump of Putty connected to the Jetson Nano running jtop.

Remember.
Login: **_jetson_**
Password: **_jetson_**

KalanaRatnayake, 2024-05-15:
Thank you and much appreciated. Hope to test this tomorrow and will leave an update.

KalanaRatnayake, 2024-05-30:
Hey All,

Ran some test on various combinations that we can use ROS on Jetson Nano using this Ubuntu image. you can find details here https://github.com/AIResearchLab/JetsonNano-ROS2.

Keep up the good work and Thanks

Qengineering, 2024-05-30:
Thanks for all the work.
I've made a link to your repo in the README.md. https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/tree/main#ros

closed 2024-05-31 by KalanaRatnayake