Qengineering/Jetson-Nano-Ubuntu-20-image #92: Not a valid link to Sync for downloading the image

pvirvani, 2024-07-23:
Hi,

I tried to download the image (both with ~10GB and ~6GB), but the links are not working.
There is server connection error with code 1000.

[image]

Qengineering, 2024-07-23:
I've the same error on multiple Sync links. It seems the server is down. Please try later.

g-r-a-e-m-e, 2024-07-29:
Currently downloading `JetsonNanoUb20_bare.img.xz` without issue. Thank y'all for having this image available.

Qengineering, 2024-07-30:
You're welcome

pvirvani, 2024-07-30:
Hi, the link is operational now, and the image was successfully downloaded.

pvirvani, 2024-07-30:
Hi, the link is operational now, and the image was successfully downloaded.

Thanks for your quick response.

closed 2024-07-30 by pvirvani