Qengineering/Jetson-Nano-Ubuntu-20-image #93: Stuck on NVIDIA screen at boot

meliheyuboglu, 2024-07-28:
I have connected the 20.04 img to the jetson nano in many ways, but it always gets stuck on the NVIDIA screen at boot. The original nvidia img works, only this custom img gets stuck on login.

Qengineering, 2024-07-29:
Sorry to hear.
It looks a hardware issue. Which Nano do you use? The 2 or 4 GB version. Or are you using the Orin Nano?
The image only works on the 4 GB Jetson Nano. Not on the Orin, nor on the 2 GB version.

meliheyuboglu, 2024-07-29:
I use the 4 GB Jetson Nano version. I have done it before for multiple times but now it is not working, I do not know whats the problem is.

meliheyuboglu, 2024-07-29:
I have been able to get Ubuntu 20 image working on my Nano, but now it fails. 
d738f1fe20088a1bdbd10e2358b512f7  My md5sum
D738F120088A1BDBD10E2358B512F7  original md5sum

Qengineering, 2024-07-29:
I see your md5sum differs from the original one. A typo? (f1fe20 vs F120)
It is strange the image has worked. The only "advise" is by starting with a fresh image all over again.
There are several files involved during booting. It seems one of them fails.
Tip: If possible use another new SD card. When it works, you may copy your software from the old one.

meliheyuboglu, 2024-07-29:
Thanks for reply, I will try and get back to you

Qengineering, 2024-07-29:
You're welcome

closed 2024-07-29 by meliheyuboglu

Khizar-BM, 2024-08-06:
Hey @meliheyuboglu , I'm also getting the same md5 as you. How did you manage to get it to work?

meliheyuboglu, 2024-08-06:
I followed the instructions under the "Split image." title.