Qengineering/Jetson-Nano-Ubuntu-20-image #94: Unable to run Jetson-io.py on Jetson Nano 2gb

Chazlazer, 2024-07-29:
Hi, this image has been working great so far on the Jetson Nano 2gb. 

However, I am having an issue enabling the spi pins on the Jetson Nano 2gb.

I was trying to use jetson-io.py to reconfigure the pins for this, however I cannot get it run. 

I think the files inside /boot are for the regular jetson nano 4gb version.

At least according to :

https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3275/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/adaptation_and_bringup_nano.html#

Says the boot files are p3449 for the 4gb version and the 2gb uses p3542

Anyone know how I can use the p3542 boot files or how to enable spi without using jetson-io.py?

abdosalem490, 2025-04-12:
@Chazlazer 
Just an irrelevant question: How did you use this image with the jetson nano 2GB? 
As per [issue#2](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/2), this image only works with the 4GB jetson nano. 
Did you use any special installation steps other than what was illustrated on the github main page?

Qengineering, 2025-04-13:
You may use this guide to upgrade from Ubuntu 18.4 to Ubuntu 20.04

closed 2025-04-13 by Qengineering