Qengineering/Jetson-Nano-Ubuntu-20-image #95: Minimum SD card requirements and image validation

g-r-a-e-m-e, 2024-07-30:
Hello! Just wanted to follow up from https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/92. The `JetsonNanoUb20_bare.img.xz` image downloaded successfully and appeared to be written to a known-good 64GB SD card using Etcher running on Pop!_OS 22.04 LTS, but it failed validation twice. While the `.xz` was ~5.6GB, Etcher indicated 32GB for the install. As the [Installation](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image?tab=readme-ov-file#installation) section of the README shows a 32GB card would suffice for the `JetsonNanoUb20_3b.img.xz` image, I wanted to recommend two things:
1. My understanding was that the `..._bare` image should not include the additional 21GB of software, as [outlined here](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image#tip). Since the `..._bare` image was more than 32GB installed, it might be worth validating the integrity of each image.
2. If the images are intact, kindly update the minimum storage requirements for an SD card compatible with either image.

I wound up using the following image from NVIDIA:
https://developer.nvidia.com/embedded/downloads#?search=4.6.1&tx=$product,jetson_nano

I manually upgraded to Ubuntu 20.04 LTS, but this came with its own set of headaches around `libopencv` and `chromium`, similar to the caveats outlined in Qengineering's instructions at https://qengineering.eu/install-ubuntu-20.04-on-jetson-nano.html.

I hope this helps others who might be in a similar situation with needing a newer OS for the Jetson Nano.

Qengineering, 2024-07-30:
Most flashing problems are related to the size of the SD card used.
If a card has only a few bytes less than the one I used, Etcher fails.
As you know, all manufacturers produce cards with slightly different sizes, even if they are all about 32 GB. They usually differ by a few bytes. I tried to use the 'smallest' card, but there are undoubtedly cards on the market that are just a little bit smaller.

If you use a 64 GB card, you will avoid all these problems. Afterwards, enlarge the partition with GParted, and you will have enough working space for your projects.

g-r-a-e-m-e, 2024-07-31:
Thank you for the reply. I apologize, I might not have been clear in my phrasing. As it turns out, I *was* using a 64GB card. I thought it was strange the `..._base` image failed to verify as the card was freshly formatted to FAT32 and had previously been running Ubuntu for a Raspberry Pi 5. I reformatted the card after the `..._base` image failures and was able to write the image from NVIDIA with no issues, suggesting the problem is unlikely to be the card itself.