Qengineering/Jetson-Nano-Ubuntu-20-image #96: JetsonNanoUb20_bare.img.xz image burn failure

shuiyihang, 2024-08-29:
Thank you very much for your contribution, but I downloaded the image twice from sync, tried using several image burners including Pi Imager and balenaEtcher (latest version), and got burn errors in between, downloaded the image using CertUtil -hashfile . \JetsonNanoUb20_bare.img.xz MD5

[image]

Qengineering, 2024-08-29:
With your MD5 correct, I think your SD card is the problem. 
SD cards larger than 128 GB can be fragile and error-prone.<br>
I could download:<br>
[image]<br>
and burn a SD without any problem.<br>
[image]

rsemihkoca, 2024-09-03:
i get same error also. md5 do not match and after balenaetcher burning it can not be extended by gparted

[image]

rishikoushikz, 2024-09-03:
I also having the same issue if anyone have the solution for this my sd card size is 64GB

Qengineering, 2024-09-03:
First and for all, there are two versions available:
- **JetsonNanoUb20_3b** (with additional DNN software)  MD5: **D738F120088A1BDBD10E2358B512F7**
- **JetsonNanoUb20_bare** (only overclocked Ubuntu OS) MD5: **DB4FC5A1B09876B37FF57F42540B3250**

@rsemihkoca, your checksum is correct. You made a typo in the second field. 

I can't reproduce your issue however.
I just downloaded the file, burn it successfully with BalenaEtcher, and insert the SD card in a Linux PC slot.
Gparted shows the following screen. 
[image]

Here, you can alter the size of the /dev/mmcblk0p1 partition without any problem once you unmounted the section.

rsemihkoca, 2024-09-03:
i resolved mine in here

https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/97#issuecomment-2326425971

i think image is working fine with all partitions but you need to make sure sd card working fine and is empty before flashing

rishikoushikz, 2024-09-04:
hay thanks man I used rp imager it's work fine for me 👌 .

yama1119, 2025-01-24:
Good day. I have an inquiry regarding the Jetson-Nano-Ubuntu-20 image. Is it compatible with the "OKdo Nano C100 Developer Kit," which is a compatible version of the "NVIDIA Jetson Nano Developer Kit B01"? The Nano C100 features different hardware.
For OS boot storage: The Nano C100 supports either eMMC or microSD, while the Jetson Nano B01 supports microSD only. For bootloader memory: The Nano C100 uses eMMC, while the Jetson Nano B01 uses SPI flash. These are the primary differences.

Qengineering, 2025-01-25:
@yama1119, I think it's not possible. Please scan for eMMC in the issues.
Tip: https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/10

yama1119, 2025-01-27:
> [@yama1119](https://github.com/yama1119), I think it's not possible. Please scan for eMMC in the issues. Tip: [#10](https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image/issues/10)

I also thought it was probably impossible, but for some reason, it feels like there’s a catch.

The Jetson Nano B01 stores its bootloader in SPI flash. The Nano C100, on the other hand, does not have SPI flash and instead stores the bootloader in eMMC. The bus assignment for the microSD card on the Jetson Nano B01 is SDMMC1, while for the Nano C100, it is SDMMC3. However, both the Nano C100 and the Jetson Nano B01 share the same size and basic structure. They consist of two boards: a system-on-module (SOM) with NVIDIA’s SoC and a carrier board with extended interfaces.

In my opinion, in the end, both have the same basic structure, and both can boot U-Boot using only a microSD card. So, I think the priority of eMMC could be overridden by prioritizing the microSD card or deleting the files in eMMC, making the Nano C100 and the Jetson Nano B01 behave the same way.

That being said, I’m trying it, but it’s not going well. (Haha)