Qengineering/Jetson-Nano-Ubuntu-20-image #97: APP is Unknown partition can not extend

rsemihkoca, 2024-09-02:
I have successfully downloaded and burned into a sd card. But on Gpart when i want to use rest of sd card it shows unknown and i can not extend it. How can we extend it ?
[image]

Qengineering, 2024-09-03:
First and for all, there are two versions available:
- **JetsonNanoUb20_3b** (with additional DNN software)  MD5: **D738F120088A1BDBD10E2358B512F7**
- **JetsonNanoUb20_bare** (only overclocked Ubuntu OS) MD5: **DB4FC5A1B09876B37FF57F42540B3250**

I can't reproduce your issue however.
I just downloaded the file, burn it successfully with BalenaEtcher, and insert the SD card in a Linux PC slot.
Gparted shows the following screen. 
[image]

Here, you can alter the size of the /dev/mmcblk0p1 partition without any problem once you unmounted the section.

rsemihkoca, 2024-09-03:
to reproduce:
JetsonNanoUb20_3b download this from link.
on windows 10 flash this image via balenaetcher to 128GB SDXC card you will get i guess 14 partition and same number of error(format this partition to recognize it in windows). 

I plug this sd card to a device running gpart on linux 20. it shows APP partition as unknown. so you can not extend it since format is unknown.

How did i resolved:

delete all partition everythin from sd card with disk manager on windows.
use raspberry pi image flasher there is link in readme.md 
flash it and when it finished it will give errors just like before but when you run gparted now it will show APP as ext4
and then i extendid now it is booting without any error.

closed 2024-09-03 by rsemihkoca

semih-karakaya, 2024-11-11:
> to reproduce: JetsonNanoUb20_3b download this from link. on windows 10 flash this image via balenaetcher to 128GB SDXC card you will get i guess 14 partition and same number of error(format this [...]

how did you extend? I guess I can't do the extend part. Did you extend all partitions to app or not? Can you be more spesific? Thank you.

Qengineering, 2024-11-12:
The image has a lot of small partitions. Leave these. 
The important partition is the large one, the one with the size of 20 GB. (In the picture above called ` /dev/mmcblk0p1`.
Now, first unmount this partition (with the right-click context menu). 
Then you can resize this partition to the full size of the card.
Last action is the clicking the green 'apply' button

semih-karakaya, 2024-11-12:
[image]
I did the gparted thing as you mentioned but problem still exist, what should I can try?

semih-karakaya, 2024-11-12:
Yes, I had problems before Gparted. Now I will install 18.04 jetpack and try to update.